BreeeZe/rpos #100: No error, but no output

ScruffySoft, 2020-12-11:
I finally managed to build rpos and get a run with no error codes. It was meant to use  GStreamer (I compiled that). However, when I run the command node rpos.js I get the following:

Read IP address 192.168.1.47 from wlan0
Manufacturer : RPOS Raspberry Pi
Model : undefined
HardwareId : 
SerialNumber : 1000000081dc05dc
FirmwareVersion : 2.0.4
Starting camera settings webserver on http://192.168.1.47:8081/
Binding DeviceService to http://192.168.1.47:8081/onvif/device_service
Binding MediaService to http://192.168.1.47:8081/onvif/media_service
Binding PTZService to http://192.168.1.47:8081/onvif/ptz_service
Binding ImagingService to http://192.168.1.47:8081/onvif/imaging_service
discovery_service started
Starting Live555 rtsp server
media_service started
imaging_service started
ptz_service started
device_service started
Opening the various links results in blank web pages, though http://192.168.1.47:8081/ does give settings. My rposConfig.json file is:

{
  "NetworkAdapters" : ["awdl0","eth0", "wlan0", "en0"], 
  "IpAddress" : "192.168.1.47",
  "ServicePort" : 8081,
  "Username" : "",
  "Password" : "",
  "CameraType" : "picam",
  "RTSPAddress" : "", "//":"Normally left blank. Used to set RTSP Server Address",
  "RTSPPort" : 443,
  "RTSPName" : "h264",
  "MulticastEnabled"  : true,
  "RTSPMulticastName" : "h264m",
  "MulticastAddress"  : "199.0.0.1",
  "MulticastPort"     : "10001",
  "RTSPServer" : 1, "RtspServerComment" : "## Select RTSP Server > 1:RPOS RTSP Server 2:V4L2 RTSP Server by mpromonet (auto selected if MulticastEnabled=true)",
  "PTZDriver" : "pan-tilt-hat", "PTZDriverComment": "## valid values are none,tenx,pelcod,visca and pan-tilt-hat",
  "PTZOutput" : "none", "PTZOutputComment": "## values are none (eg Tenx), serial and tcp",
  "PTZSerialPort" : "/dev/ttyUSB0",
  "PTZSerialPortSettings" : { "baudRate":2400, "dataBits":8, "parity":"none", "stopBits":1 },
  "PTZOutputURL": "224.0.0.1:9999",
  "PTZCameraAddress": 1,
  "DeviceInformation" : {
    "Manufacturer" : "Raspberry Pi",
    "Model" : "4 B",
    "HardwareId" : ""
  },
  "logLevel" : 3, "logLevelComment": "## LogLevels are > 1:Error 2:Warning 3:Info 4:Debug",
  "logSoapCalls" : false
}

I use port 443 because the default 8554 port gave an ioctl error. 
Can you help me with the following questions:
Why is it not using Gstreamer and how do I get it to?
How do I set the resolution of the camera image?
Why don't I get anything on those service pages?
 Thanks

EduardoRonchi, 2021-04-23:
Hello, did you managed to make it work? I am facing the same erorrs that you did. My issue: https://github.com/BreeeZe/rpos/issues/106

ScruffySoft, 2021-04-27:
No, I haven't found a solution. I've pretty much accepted that I cannot have the zoom camera ability and stream the camera output to FB or even VLC at the same time. So it's PT not PTZ until someone makes some part of the software work.