BreeeZe/rpos #103: PTZ control

osvikvi, 2021-02-05:
Hello folks, 
I'm a bit of a newbie when it comes to raspberry but I'm starting to get the hang of it. . But now I've come across some issues I can't seem to get resolved by myself. 
I managed to install rpos succesfully and can connect to the VISCAcamera via a windows server running NetCamStudio, next I'm trying to setup PTZ, but I keep on failing to get control over my Tandberg PrecisionHD PTZ camera (it basically is a PTZ camera with HDMI video output and a serial port for VISCA control.)
I have tried on both a USB to serial adapter as on the on board  serial port of the raspi, but the camera doesn't move. Whatever command I give from my windows server, nothing happens. I've enabled logging to the max in RposConfig.jason which threw me these errors at each and every move.
`<s:Envelope 
xmlns:s="http://www.w3.org/2003/05/soap-envelope" 
xmlns:a="http://www.w3.org/2005/08/addressing">
<s:Header>
<Security xmlns="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-wssecurity-secext-1.0.xsd">
	<UsernameToken>
		<Username>admin</Username>
		<Password Type="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-username-token-profile-1.0#PasswordDigest">+h2LwGa4VmbyhbANVrgg2aTuvQk=</Password>
		<Nonce EncodingType="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-soap-message-security-1.0#Base64Binary">iRekysAorcEf63JciDJXbQ==</Nonce>
		<Created xmlns="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-wssecurity-utility-1.0.xsd">2021-02-04T09:27:08.752Z</Created>
	</UsernameToken>
</Security>
<a:To s:mustUnderstand="1">http://192.168.123.234:8081/onvif/device_service</a:To>
</s:Header>
	<s:Body xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:xsd="http://www.w3.org/2001/XMLSchema"><GetConfigurations xmlns="http://www.onvif.org/ver20/ptz/wsdl"/></s:Body>
	</s:Envelope>
	DeviceService - Calltype : error, Data : TypeError: Cannot read property 'output' of undefined
    at SAXParser.p.onopentag (/home/pi/rpos/node_modules/soap/lib/wsdl.js:1255:41)
    at emit (/home/pi/rpos/node_modules/sax/lib/sax.js:624:35)
    at emitNode (/home/pi/rpos/node_modules/sax/lib/sax.js:629:5)
    at openTag (/home/pi/rpos/node_modules/sax/lib/sax.js:825:5)
    at SAXParser.write (/home/pi/rpos/node_modules/sax/lib/sax.js:1292:13)
    at WSDL.xmlToObject (/home/pi/rpos/node_modules/soap/lib/wsdl.js:1381:5)
    at Server._process (/home/pi/rpos/node_modules/soap/lib/server.js:151:21)
    at IncomingMessage.<anonymous> (/home/pi/rpos/node_modules/soap/lib/server.js:124:14)
    at emitNone (events.js:111:20)
    at IncomingMessage.emit (events.js:208:7)
web request received : /onvif/PTZ`

I've tried changing all parameters without success, tried changing serial ports... I did enabled serial port in the raspi-config as instructed on several pages, tried with minicom to test the serial port, etcetera but no luck... I'm not even sure if the "outpu"-error is related to the serial port.

can anyone shed some light on the matter? Or tell me which background info I failed to provide in order to solve the problem ;)

THanks for your time

osvikvi, 2021-02-14:
bump...

Is it really so? does no-one knows why these errors are thrown? or how to resolve them.
I've been looking around in the mentioned *.js files but with my limited knowledge can't seem to find anything that makes sense to me. (I'l attach them below, maybe someone with more knowledge can have a look) 

[sax.js.txt](https://github.com/BreeeZe/rpos/files/5976108/sax.js.txt)
[server.js.txt](https://github.com/BreeeZe/rpos/files/5976109/server.js.txt)
[wsdl.js.txt](https://github.com/BreeeZe/rpos/files/5976110/wsdl.js.txt)

uglurass, 2021-02-14:
i cant open them in google phone thats weird

osvikvi, 2021-02-15:
takk for having a look,

I had to change file extension in order to be able to upload *.js files here, I don't know what you mean by "Google phone" but from my android device through the e-maillink sent by github, I can just open these *.js.txt files without issue. 
Rigtig underligt... ;)

uglurass, 2021-02-15:
google isisnt be iintrusif tha blew up when they collect your routr that is
visit the responsible

osvikvi, 2021-02-15:
Thanks for your feedback, unfortunately I don't quite understand what you said. Would you mind to elaborate. I don't think that Google or my Router have anything to do with this, or do they?

osvikvi, 2021-02-16:
@BreeeZe @RogerHardiman  @caspermeijn 
please excuse me for my persistence, I'm keen to get PTZ control to work from RPOS.
 In the meantime I can Confirm my serial port  /dev/ttyUSB0 is working, I can move the camera sending [0x81,0x01,0x06,0x01,0x01,0x01,0x01,0x03,0xff] through a py script. but I can't get RPOS to do the same. the errors remain exactly the same whatever I do or change. 

When I browse to: [(http://192.168.123.234:8081/onvif/ptz_service)](http://192.168.123.234:8081/onvif/ptz_service)
RPOS responds with:
> web request received : /onvif/
> PTZService - Calltype : info, Data : Handling GET on /onvif/ptz_service
but the browser page remains blank...

Would anyone please have any suggestion? 
bedankt

osvikvi, 2021-02-19:
it seems that my persistence payed off. re-installing the entire shizzle brought me a step closer, I can now finally connect AND control using ODM. :)
Strangely enough whilst ODM recognizes just a single profile , the webcam server application I am using ([NetCam Studio](https://netcamstudio.com/)) cannot discover a valid profile. :(
[image]

also worth mentioning is that SOAP responses from ODM -querries are much different than when coming from NCS.

I've also asked the dev's over there for assistance, let's see if we can get any wiser...

RogerHardiman, 2021-02-21:
Hi
It is good to know you got the VISCA control working.
I have downloaded NCS and will have a quick look

RogerHardiman, 2021-02-21:
The problem is a bug in NCS.
They are sending GetProfiles requests to the wrong ONVIF URL.
They are sending GetProfiles to a URL ending /onvif/device_service

But GetProfiles is a **Media** command and not a **Device** command and needs to be sent to the URL given by the GetCapabilities command and the **Media XADDR**.   

If NCS fix their bug then I expect it will work fine.

osvikvi, 2021-02-22:
Thanks a lot, I was expecting something like that.... I'll ask them if they recognize this, and if they can provide a fix.

Thanks again, that's really been most helpfull. Now that I have your attention, I'm still having dificulties Making Rpos autostart at boot. Do you have a suggestion?
It goes without saying that If there's I can do anything for you, beta testing, cleanup or anything just let me know. 

Vik

RogerHardiman, 2021-02-22:
I just add something like
```
cd /home/pi/rpos
node rpos.js &
```

to /etc/rc.local

Thanks
Roger

closed 2021-03-04 by osvikvi