BreeeZe/rpos #106: Error rtspServer: VIDIOC_REQBUFS: Inappropriate ioctl for device

EduardoRonchi, 2021-04-23:
Hello, I am trying to run thin project with a Raspberry Pi 3B+ and a Raspbiberry Camera Module v1.

I followed all the steps and when I run $ node rpos.js i get the following output (image below):
pi@raspberrypi:~/rpos $ node rpos.js
Read IP address 192.168.0.10 from wlan0
Manufacturer : RPOS Raspberry Pi
Model : Model_B_PI_3
HardwareId : 
SerialNumber : 000000000e625583
FirmwareVersion : 2.0.4
Starting camera settings webserver on http://192.168.0.10:8081/
Binding DeviceService to http://192.168.0.10:8081/onvif/device_service
Binding MediaService to http://192.168.0.10:8081/onvif/media_service
Binding PTZService to http://192.168.0.10:8081/onvif/ptz_service
Binding ImagingService to http://192.168.0.10:8081/onvif/imaging_service
discovery_service started
imaging_service started
ptz_service started
device_service started
Starting Live555 rtsp server
media_service started
rtspServer: VIDIOC_REQBUFS: Inappropriate ioctl for device
VIDIOC_STREAMOFF: Inappropriate ioctl for device
VIDIOC_REQBUFS: Inappropriate ioctl for device

rtspServer: VIDIOC_REQBUFS: Inappropriate ioctl for device
VIDIOC_STREAMOFF: Inappropriate ioctl for device
VIDIOC_REQBUFS: Inappropriate ioctl for device

[image]

I am using OPTION 2: USING MPROMONET RTSP SERVER in the config file. I also tried option 3 compiling from source, but without success. Any help?

Thanks!

EduardoRonchi, 2021-04-25:
Update: reinstalled node v8 and now the system hangs after device_service started and do not give me an RSTP. Also changed the to a raspberry pi 4 to see if anything changed, but the situation is the same.

pi@raspberrypi:~/rpos $ node rpos.js
Read IP address 192.168.0.10 from wlan0
Manufacturer : RPOS Raspberry Pi
Model : Model_B_PI_3
HardwareId :
SerialNumber : 000000000e625583
FirmwareVersion : 2.0.4
Starting camera settings webserver on http://192.168.0.10:8081/
Binding DeviceService to http://192.168.0.10:8081/onvif/device_service
Binding MediaService to http://192.168.0.10:8081/onvif/media_service
Binding PTZService to http://192.168.0.10:8081/onvif/ptz_service
Binding ImagingService to http://192.168.0.10:8081/onvif/imaging_service
discovery_service started
imaging_service started
ptz_service started
device_service started
Starting Live555 rtsp server
media_service started

RogerHardiman, 2021-05-15:
I will give this a try on a Pi 3 which has a Pi Camera.
We pull in the rtsp server from MPromonet so I guess there is an issue with Kernel Versions and his RTSP server

EduardoRonchi, 2021-05-15:
Update: I made it work with the raspiberry cam v1. I just reinstalled node with nvm to the 8.x version and it worked. I thin I was using the 10.x version when the error happened.

closed 2021-05-19 by EduardoRonchi