BreeeZe/rpos #107: I'm stuck on STEP 3

threeeye, 2021-05-03:
In step 3 I get stuck with the following command:
> npm install

The output is (long):

```
pi@raspberrypi:~/rpos $ npm install
npm ERR! code 1
npm ERR! path /home/pi/rpos/node_modules/serialport
npm ERR! command failed
npm ERR! command sh -c prebuild-install || node-gyp rebuild
npm ERR! make: Entering directory '/home/pi/rpos/node_modules/serialport/build'
npm ERR!   CXX(target) Release/obj.target/serialport/src/serialport.o
npm ERR! make: Leaving directory '/home/pi/rpos/node_modules/serialport/build'
npm ERR! gyp info it worked if it ends with ok
npm ERR! gyp info using node-gyp@7.1.2
npm ERR! gyp info using node@16.0.0 | linux | arm
npm ERR! gyp info find Python using Python version 3.7.3 found at "/usr/bin/python3"
npm ERR! (node:897) [DEP0150] DeprecationWarning: Setting process.config is deprecated. In the future the property will be read-only.
npm ERR! (Use `node --trace-deprecation ...` to show where the warning was created)
npm ERR! gyp info spawn /usr/bin/python3
npm ERR! gyp info spawn args [
npm ERR! gyp info spawn args   '/home/pi/.nvm/versions/node/v16.0.0/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py',
npm ERR! gyp info spawn args   'binding.gyp',
npm ERR! gyp info spawn args   '-f',
npm ERR! gyp info spawn args   'make',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/home/pi/rpos/node_modules/serialport/build/config.gypi',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/home/pi/.nvm/versions/node/v16.0.0/lib/node_modules/npm/node_modules/node-gyp/addon.gypi',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/home/pi/.cache/node-gyp/16.0.0/include/node/common.gypi',
npm ERR! gyp info spawn args   '-Dlibrary=shared_library',
npm ERR! gyp info spawn args   '-Dvisibility=default',
npm ERR! gyp info spawn args   '-Dnode_root_dir=/home/pi/.cache/node-gyp/16.0.0',
npm ERR! gyp info spawn args   '-Dnode_gyp_dir=/home/pi/.nvm/versions/node/v16.0.0/lib/node_modules/npm/node_modules/node-gyp',
npm ERR! gyp info spawn args   '-Dnode_lib_file=/home/pi/.cache/node-gyp/16.0.0/<(target_arch)/node.lib',
npm ERR! gyp info spawn args   '-Dmodule_root_dir=/home/pi/rpos/node_modules/serialport',
npm ERR! gyp info spawn args   '-Dnode_engine=v8',
npm ERR! gyp info spawn args   '--depth=.',
npm ERR! gyp info spawn args   '--no-parallel',
npm ERR! gyp info spawn args   '--generator-output',
npm ERR! gyp info spawn args   'build',
npm ERR! gyp info spawn args   '-Goutput_dir=.'
npm ERR! gyp info spawn args ]
npm ERR! gyp info spawn make
npm ERR! gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
npm ERR! In file included from ../src/./serialport.h:6,
npm ERR!                  from ../src/serialport.cpp:1:
npm ERR! ../../nan/nan.h: In function ‘void Nan::AsyncQueueWorker(Nan::AsyncWorker*)’:
npm ERR! ../../nan/nan.h:2298:62: warning: cast between incompatible function types from ‘void (*)(uv_work_t*)’ {aka ‘void (*)(uv_work_s*)’} to ‘uv_after_work_cb’ {aka ‘void (*)(uv_work_s*, int)’} [-Wcast-function-type]
npm ERR!      , reinterpret_cast<uv_after_work_cb>(AsyncExecuteComplete)
npm ERR!                                                               ^
npm ERR! ../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Open(Nan::NAN_METHOD_ARGS_TYPE)’:
npm ERR! ../src/serialport.cpp:41:48: error: no matching function for call to ‘v8::Value::ToString()’
npm ERR!    v8::String::Utf8Value path(info[0]->ToString());
npm ERR!                                                 ^
npm ERR! In file included from /home/pi/.cache/node-gyp/16.0.0/include/node/node.h:63,
npm ERR!                  from ../../nan/nan.h:54,
npm ERR!                  from ../src/./serialport.h:6,
npm ERR!                  from ../src/serialport.cpp:1:
npm ERR! /home/pi/.cache/node-gyp/16.0.0/include/node/v8.h:3019:44: note: candidate: ‘v8::MaybeLocal<v8::String> v8::Value::ToString(v8::Local<v8::Context>) const’
npm ERR!    V8_WARN_UNUSED_RESULT MaybeLocal<String> ToString(
npm ERR!                                             ^~~~~~~~
npm ERR! /home/pi/.cache/node-gyp/16.0.0/include/node/v8.h:3019:44: note:   candidate expects 1 argument, 0 provided
npm ERR! ../src/serialport.cpp:48:53: error: no matching function for call to ‘v8::Value::ToObject()’
npm ERR!    v8::Local<v8::Object> options = info[1]->ToObject();
npm ERR!                                                      ^
npm ERR! In file included from /home/pi/.cache/node-gyp/16.0.0/include/node/node.h:63,
npm ERR!                  from ../../nan/nan.h:54,
[... 232 lines cut ...]
npm ERR!        (node::addon_register_func) (regfunc),                          \
npm ERR!                                     ^~~~~~~
npm ERR! ../src/serialport.cpp:485:1: note: in expansion of macro ‘NODE_MODULE’
npm ERR!  NODE_MODULE(serialport, init);
npm ERR!  ^~~~~~~~~~~
npm ERR! make: *** [serialport.target.mk:114: Release/obj.target/serialport/src/serialport.o] Error 1
npm ERR! gyp ERR! build error
npm ERR! gyp ERR! stack Error: `make` failed with exit code: 2
npm ERR! gyp ERR! stack     at ChildProcess.onExit (/home/pi/.nvm/versions/node/v16.0.0/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)
npm ERR! gyp ERR! stack     at ChildProcess.emit (node:events:365:28)
npm ERR! gyp ERR! stack     at Process.ChildProcess._handle.onexit (node:internal/child_process:290:12)
npm ERR! gyp ERR! System Linux 5.10.17-v7+
npm ERR! gyp ERR! command "/home/pi/.nvm/versions/node/v16.0.0/bin/node" "/home/pi/.nvm/versions/node/v16.0.0/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
npm ERR! gyp ERR! cwd /home/pi/rpos/node_modules/serialport
npm ERR! gyp ERR! node -v v16.0.0
npm ERR! gyp ERR! node-gyp -v v7.1.2
npm ERR! gyp ERR! not ok
npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2021-05-03T18_13_46_300Z-debug.log
pi@raspberrypi:~/rpos $
```
I'm attaching the full log
[2021-05-03T18_13_46_300Z-debug.log](https://github.com/BreeeZe/rpos/files/6416866/2021-05-03T18_13_46_300Z-debug.log)

Thanks

SirDifferential, 2021-05-14:
This happens due to [serialport](https://www.npmjs.com/package/serialport) 6.0.4 not working with nodejs 12. Updating serialport to 9.0.7 fixes the problem.

So:

* Delete `package-lock.json`
* Edit `package.json` and change `"serialport": "^6.0.4",`  to `"serialport": "^9.0.7",`
* Run `npm install`

I'd open a pull request, but this essentially re-creates `package-json.lock` and I'm not sure how nodejs based projects handle auditing / vetting changes to dependencies at this scale.

On similar note, npm tells me there are 22 vulnerabilities, out of which 12 are high priority. I suppose these should be looked at?

threeeye, 2021-05-14:
Thanks for the reply.
This fixed that issue.
But now I get this on Step 4.1.a:
```
pi@raspberrypi:/rpos $ npx gulp
{ AssertionError [ERR_ASSERTION]: Task function must be specified
    at Gulp.set [as _setTask] (/home/pi/rpos/node_modules/undertaker/lib/set-task.js:10:3)
    at Gulp.task (/home/pi/rpos/node_modules/undertaker/lib/task.js:13:8)
    at Object.<anonymous> (/home/pi/rpos/gulpfile.js:37:6)
    at Module._compile (internal/modules/cjs/loader.js:778:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)
    at Module.require (internal/modules/cjs/loader.js:692:17)
    at require (internal/modules/cjs/helpers.js:25:18)
  generatedMessage: false,
  name: 'AssertionError [ERR_ASSERTION]',
  code: 'ERR_ASSERTION',
  actual: false,
  expected: true,
  operator: '==' }
pi@raspberrypi:~/rpos $
```

RogerHardiman, 2021-05-15:
The problem is that the master branch of RPOS was written for Node JS Version 8 and 

There is a line in the readme that says "NOTE: Node.js Version 6.x and 8.x have been tested with RPOS. Only a small amount of testing has been done with Node v10."

If you roll back your version of Node JS then you will be OK and Serial Port version 6 will work.

But last weekend I updated the project to use Node v12 which uses the new Serial Port code.
Node12 changes also use Gulp Version 4 (as Gulp v3 is not compatible with Node 12) which changed the way Gulp Tasks operate.

So I wonder if you have the wrong version of node, or have not re-run ```npm install``` after pulling in my most recent commits, or maybe I have forgotten to commit a local change.

Karlo318, 2022-02-28:
Hello,

I am facing a similar problem in STEP 3. I am trying to create ONVIF camera using Raspberry Pi Zero W and official camera module v2. After running npm install I get the following error:

>pi@still2:~/rpos $ npm install
> @serialport/bindings@9.1.0 install /home/pi/rpos/node_modules/@serialport/bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild
>prebuild-install WARN install No prebuilt binaries found (target=11.15.0 runtime=node arch=arm libc= platform=linux)
>gyp ERR! configure error
>gyp ERR! stack Error: Command failed: /usr/bin/python -c import sys; print "%s.%s.%s" % sys.version_info[:3];
>gyp ERR! stack   File "<string>", line 1
>gyp ERR! stack     import sys; print "%s.%s.%s" % sys.version_info[:3];
gyp ERR! stack                       ^
gyp ERR! stack SyntaxError: invalid syntax
gyp ERR! stack
gyp ERR! stack     at ChildProcess.exithandler (child_process.js:299:12)
gyp ERR! stack     at ChildProcess.emit (events.js:193:13)
gyp ERR! stack     at maybeClose (internal/child_process.js:999:16)
gyp ERR! stack     at Socket.stream.socket.on (internal/child_process.js:403:11)
gyp ERR! stack     at Socket.emit (events.js:193:13)
gyp ERR! stack     at Pipe._handle.close (net.js:614:12)
gyp ERR! System Linux 5.10.92+
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild                            "
gyp ERR! cwd /home/pi/rpos/node_modules/@serialport/bindings
gyp ERR! node -v v11.15.0
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.13 (node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.13: wanted {"os":"darwin","arch":"a                            ny"} (current: {"os":"linux","arch":"arm"})
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! @serialport/bindings@9.1.0 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the @serialport/bindings@9.1.0 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2022-02-28T11_44_02_523Z-debug.log

Do you have any idea how to solve this problem? Any help is appreciated!