BreeeZe/rpos #109: Feature request: output PTZ to GPIO

AvrumFeldman, 2021-05-30:
Hi,

This project is something that I've looked everywhere for, so thank you.

I found out that the yt-260 can be controlled using the GPIO pins of an RPI. I was wondering is it is possible it implement  PT~~Z~~ output to GPIO pins?

I was thinking a section in the config file where we can map the pins to direction e.g. up = 9, down = 10, right = 11 etc...)

I'm extremely raw in JS/TS but with some pointers I can try to to implement it myself.

RogerHardiman, 2021-05-30:
The ONVIF PTZ commands all get passed to a generic "PTZ driver" in the libs folder.
This can send out actual pan and tilt commands in lots of different formats, it can send commands to a USB Pan/Tilt unit (a rocket launcher), in the CCTV standard Pelco D and Pelco P format, in the Sony VISCA format and it drives to different Pi Servo Pan/Tilt units (one from Waveshare, one from Pimoroni)

To output commands to different pins would be simple to do and I've taken a quick look at the link you posted.

rjsdotorg, 2021-06-28:
> Hi,
>
> This project is something that I've looked everywhere for, so thank you.
>
> I found out that the [yt-260 can be controlled using the GPIO pins of an [...]

I'm working on this as well, so we can collaborate: https://github.com/BreeeZe/rpos/issues/110#issuecomment-866593248 is where Roger points out the Python code to work on.
We are using Sparkfun stepper driver controlled by Pi or Arduino GPIOs (with limit switches).

For instance, part of the Arduino code looks like this
```
void RotateStepper(int rot, int steps, int _delay, bool stop_if_parked)  {
  if(_delay<1) {
    // require min delay
    _delay = 1;
  }
  //Pull enable pin low to allow motor control
  digitalWrite(EN, LOW);
  //move wiper out across
  digitalWrite(dir1, rot);
  int y;
  for (y = 0; y < steps; y++) {
    //Trigger one step
    digitalWrite(stp1, HIGH);
    digitalWrite(stp2, HIGH);
    // need some delay
    delay(_delay);
    //Pull step pin low so it can be triggered again
    digitalWrite(stp1, LOW);
    digitalWrite(stp2, LOW);
    delay(_delay);
    if (stop_if_parked && digitalRead(PARK1) == LOW)  {
      digitalWrite(LED_PARK, HIGH);
      //Serial.print("PARKED!");
      //Serial.println(digitalRead(PARK1));
      delay(100);
      break;
    }
    if (y % 200 == 0) {
      ToggleLED(LED_STEP);
      if (rot == 0) {
        Serial.print(">");
      } else  {
        Serial.print("<");
      }
      //yield;
    }
  }
  //Pull enable pin high to idle motor controler
  digitalWrite(dir1, HIGH);
  digitalWrite(LED_STEP, LOW);
}
```
and is being being translated to Python/Pi.

RogerHardiman, 2022-03-12:
I'll close this issue report down. RPOS can now pass high level commands for PTZ over a socket interface for someone to write their own stepper motor control code.

For the specific Pan/Tilt platform mentioned, as I don't have access to one I am not able to add any software for it, but would accept contributions

closed 2022-03-12 by RogerHardiman