BreeeZe/rpos #114: rpos as ONVIF Proxy Server

diazepam69, 2021-07-13:
Hello all

I'm trying to use my old IP Camera (which doesn't have Onvif support) on my new NVR, and I saw this project and sounds promising to me.

Basically, can I use this setup with rpos? setting in configuration for use and external RTSP address (instead of usb or picam) and "convert" to Onvif compliant? (my NVR will connect direct to rpos server)

Thanks!

RogerHardiman, 2021-07-14:
Hi
Yes you can do this. RPOS handles the ONVIF messages and supplies the RTSP URL to the NVR.
The NVR then goes and gets the video from the camera.

diazepam69, 2021-07-14:
Good. But in rpos config, which camera type should I put? ( "CameraType" parameter)

yamith9, 2021-09-04:
> Good. But in rpos config, which camera type should I put? ( "CameraType" parameter)

Were You able to resolve this? I'm trying to do they same but i'm stuck here

pistonov, 2021-09-07:
The same question

yamith9, 2021-09-07:
> The same question

I found hownto solve it. Un RSTPServer male it equal to 0. Even if the doc says only values are 1 to 3, 0 is a valid value

pistonov, 2021-09-08:
What about "CameraType" parameter)? I got error "Cannot open device /dev/video0, exiting."

RogerHardiman, 2021-09-08:
@yamith9 is correct. Set the paraemter called RTSPServer to 0 in the config file

When the RTSPServer value in the config file is set to 0, the CameraType parameter is ignored.
The value of 0 was included in a commit 2d8fdee but the config file instructions were never updated

RogerHardiman, 2021-09-08:
I have added a Proxy sample file. Can you all check it please and let me know if you need any changes

pistonov, 2021-09-08:
I'v got same error "Cannot open device /dev/video0, exiting."

# node rpos.js
Read IP address 10.54.0.100 from eth0
Manufacturer : RPOS Proxy
Model : PROXY
HardwareId : 
SerialNumber : 000000002f85c92e
FirmwareVersion : 1.0.0
Starting camera settings webserver on http://10.54.0.100:8081/
Cannot open device /dev/video0, exiting.
Cannot open device /dev/video0, exiting.
Cannot open device /dev/video0, exiting.
Cannot open device /dev/video0, exiting.
Cannot open device /dev/video0, exiting.
Could not retrieve Controlvalue 'brightness'
Could not retrieve Controlvalue 'contrast'
Could not retrieve Controlvalue 'saturation'
Could not retrieve Controlvalue 'red_balance'
Could not retrieve Controlvalue 'blue_balance'
Could not retrieve Controlvalue 'horizontal_flip'
Could not retrieve Controlvalue 'vertical_flip'
Could not retrieve Controlvalue 'power_line_frequency'
Could not retrieve Controlvalue 'sharpness'
Could not retrieve Controlvalue 'color_effects'
Could not retrieve Controlvalue 'rotate'
Could not retrieve Controlvalue 'color_effects_cbcr'
Could not retrieve Controlvalue 'video_bitrate_mode'
Could not retrieve Controlvalue 'video_bitrate'
Could not retrieve Controlvalue 'repeat_sequence_header'
Could not retrieve Controlvalue 'h264_i_frame_period'
Could not retrieve Controlvalue 'h264_level'
Could not retrieve Controlvalue 'h264_profile'
Could not retrieve Controlvalue 'auto_exposure'
Could not retrieve Controlvalue 'exposure_time_absolute'
Could not retrieve Controlvalue 'exposure_dynamic_framerate'
Could not retrieve Controlvalue 'auto_exposure_bias'
Could not retrieve Controlvalue 'white_balance_auto_preset'
Could not retrieve Controlvalue 'image_stabilization'
Could not retrieve Controlvalue 'iso_sensitivity'
Could not retrieve Controlvalue 'exposure_metering_mode'
Could not retrieve Controlvalue 'scene_mode'
Could not retrieve Controlvalue 'compression_quality'
Binding DeviceService to http://10.54.0.100:8081/onvif/device_service
Binding MediaService to http://10.54.0.100:8081/onvif/media_service
Binding PTZService to http://10.54.0.100:8081/onvif/ptz_service
Binding ImagingService to http://10.54.0.100:8081/onvif/imaging_service
discovery_service started
Uncaught Exception... : TypeError [ERR_INVALID_ARG_TYPE]: The "path" argument must be of type string. Received undefined
    at validateString (internal/validators.js:120:11)
    at Object.dirname (path.js:1128:5)
    at WSDL._processNextInclude (/home/rpos/node_modules/soap/lib/wsdl/index.js:1084:49)
    at WSDL.processIncludes (/home/rpos/node_modules/soap/lib/wsdl/index.js:138:14)
    at /home/rpos/node_modules/soap/lib/wsdl/index.js:74:19
    at processTicksAndRejections (internal/process/task_queues.js:79:11)

RogerHardiman, 2021-09-08:
There will be a lot of errors/warnings we can ignore - like all the "cound not retrieve Controlvalue" ones.
At some point I'll have to pop in an if-statement there to skip over them, but you can ignore them.

However you have something more than that -- an uncaught exception.
I wonder if I have a problem in the sameple json file. I never tested it, just wrote it and added it to github without testing

pistonov, 2021-09-08:
I fixed it by reinstalling the soup module. Now i haven't any exceptions.

diazepam69, 2021-09-19:
Everything worked well with your tips guys, thanks!

GeniusOnline, 2021-11-24:
> Cannot open device /dev/video0, exiting.
> Cannot open device /dev/video0, exiting.
> Cannot open device /dev/video0, exiting.
> Cannot open device /dev/video0, exiting.
> Cannot open device /dev/video0, exiting.

these can be fixed by adding --device /dev/video* to /lib/v4l2ctl.js 

[image]

Softfruits2022, 2022-06-22:
Hi, where do I put the external RTSP-Address/User/Port etc. of my old rtsp-camera? I can't find a place in the rpos config file, or am i missing something. I want to get my old RTSP-Cam running on rpos to use it as ONVIF Device. Thanks for help !

Softfruits2022, 2022-06-22:
Thanks a lot ! I did that and on the Website it says on the top:
Video Stream: rtsp://username:password@deviceIPaddress:554/live1

Something wrong with my settings, the camera has no user and pass?:
{
  "NetworkAdapters" : ["awdl0","eth0", "wlan0", "en0"], 
  "IpAddress" : "192.168.10.99",
  "ServicePort" : 8081,
  "Username" : "cam",
  "Password" : "cam",
  "CameraType" : "picam",
  "RTSPAddress" : "192.168.10.107",
  "RTSPPort" : 554,
  "RTSPName" : "live1",
  "MulticastEnabled"  : false,
  "RTSPMulticastName" : "h264m",
  "MulticastAddress"  : "224.0.0.1",
  "MulticastPort"     : "10001",
  "RTSPServer" : 0, "RtspServerComment" : "## Select RTSP Server > 0:Use RTSP Server from 'RTSPAddress:' 1:RPOS RTSP Server 2:V4L2 RTSP Server by mpromonet (auto selected if MulticastEnabled=true) 3:GStreamer RTSP Server",
  "PTZDriver" : "none", "PTZDriverComment": "## valid values are none,tenx,pelcod,visca and pan-tilt-hat",
  "PTZOutput" : "none", "PTZOutputComment": "## values are none (eg Tenx), serial and tcp",
  "PTZSerialPort" : "/dev/ttyUSB0",
  "PTZSerialPortSettings" : { "baudRate":2400, "dataBits":8, "parity":"none", "stopBits":1 },
  "PTZOutputURL": "127.0.0.1:9999",
  "PTZCameraAddress": 1,
  "DeviceInformation" : {
    "Manufacturer" : "RPOS Proxy",
    "Model" : "PROXY",
    "HardwareId" : ""
  },
  "logLevel" : 3, "logLevelComment": "## LogLevels are > 1:Error 2:Warning 3:Info 4:Debug",
  "logSoapCalls" : false
}

Softfruits2022, 2022-06-23:
Thanks, it works fine now! It never shows the settings for RTSPAddress for security reasons!

philnagel, 2023-08-05:
Hi! Hopefully it's OK to continue here, let me know if I should open a new issue. I am trying to set this up as a ONVIF proxy only. My understanding is that it will listen for ONVIF queries and than point the client to the correct stream URL, which is a different IP (camera ONVIF not compatible with my NVR is the reason for this setup).
But it seems that it is instead returning the IP of the machine that is running RPOS instead of the one configured for the camera on my config for the RTSP stream, at least that's how I interpret the output. My client is unable to connect to the RTSP stream, but is is reading the ONVIF config fine (using onvier on Android for testing).

Do I have this configured incorrectly or am I misunderstanding what should be happening?

Thank you!

RPOS output:

```
cat: /proc/device-tree/model: No such file or directory
Failed to read serial : Cannot read property '1' of null
Using IP address from config: 192.168.15.29
Manufacturer : RPOS Proxy
Model : PROXY
HardwareId :
SerialNumber : 000000000
FirmwareVersion : 2.1.0
Starting camera settings webserver on http://192.168.15.29:8081/
/bin/sh: 1: v4l2-ctl: not found
/bin/sh: 1: v4l2-ctl: not found
/bin/sh: 1: v4l2-ctl: not found
/bin/sh: 1: v4l2-ctl: not found
/bin/sh: 1: v4l2-ctl: not found
Could not retrieve Controlvalue 'brightness'
Could not retrieve Controlvalue 'contrast'
Could not retrieve Controlvalue 'saturation'
Could not retrieve Controlvalue 'red_balance'
Could not retrieve Controlvalue 'blue_balance'
Could not retrieve Controlvalue 'horizontal_flip'
Could not retrieve Controlvalue 'vertical_flip'
Could not retrieve Controlvalue 'power_line_frequency'
Could not retrieve Controlvalue 'sharpness'
Could not retrieve Controlvalue 'color_effects'
Could not retrieve Controlvalue 'rotate'
Could not retrieve Controlvalue 'color_effects_cbcr'
Could not retrieve Controlvalue 'video_bitrate_mode'
Could not retrieve Controlvalue 'video_bitrate'
Could not retrieve Controlvalue 'repeat_sequence_header'
Could not retrieve Controlvalue 'h264_i_frame_period'
Could not retrieve Controlvalue 'h264_level'
Could not retrieve Controlvalue 'h264_profile'
Could not retrieve Controlvalue 'auto_exposure'
Could not retrieve Controlvalue 'exposure_time_absolute'
Could not retrieve Controlvalue 'exposure_dynamic_framerate'
Could not retrieve Controlvalue 'auto_exposure_bias'
Could not retrieve Controlvalue 'white_balance_auto_preset'
Could not retrieve Controlvalue 'image_stabilization'
Could not retrieve Controlvalue 'iso_sensitivity'
Could not retrieve Controlvalue 'exposure_metering_mode'
Could not retrieve Controlvalue 'scene_mode'
Could not retrieve Controlvalue 'compression_quality'
Binding DeviceService to http://192.168.15.29:8081/onvif/device_service
Binding MediaService to http://192.168.15.29:8081/onvif/media_service
Binding PTZService to http://192.168.15.29:8081/onvif/ptz_service
Binding ImagingService to http://192.168.15.29:8081/onvif/imaging_service
discovery_service started
Starting rtsp server
media_service started
imaging_service started
device_service started
ptz_service started
(node:682000) [DEP0005] DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.
ffmpeg - starting
ffmpeg - finished
ffmpeg exec error: Error: Command failed: ffmpeg -fflags nobuffer -probesize 256 -rtsp_transport tcp -i rtsp://127.0.0.1:554/1/h264major -vframes 1  -r 1 -s 640x360 -y /dev/shm/snapshot.jpg
/bin/sh: 1: ffmpeg: not found

    at ChildProcess.exithandler (child_process.js:308:12)
    at ChildProcess.emit (events.js:314:20)
    at maybeClose (internal/child_process.js:1022:16)
    at Socket.<anonymous> (internal/child_process.js:444:11)
    at Socket.emit (events.js:314:20)
    at Pipe.<anonymous> (net.js:675:12) {
  killed: false,
  code: 127,
  signal: null,
  cmd: 'ffmpeg -fflags nobuffer -probesize 256 -rtsp_transport tcp -i rtsp://127.0.0.1:554/1/h264major -vframes 1  -r 1 -s 640x360 -y /dev/shm/snapshot.jpg'
}
```

Config:

```
{
  "NetworkAdapters" : ["awdl0","eth0", "wlan0", "en0", "enp1s0"],
  "IpAddress": "192.168.15.29",
  "ServicePort" : 8081,
  "Username" : "admin",
  "Password" : "admin",
  "CameraType" : "picam",
  "RTSPAddress" : "admin:admin@192.168.1.251", "//":"Normally left blank. Used to set RTSP Server Address",
  "RTSPPort" : 554,
  "RTSPName" : "1/h264major",
  "MulticastEnabled"  : false,
  "RTSPMulticastName" : "h264m",
  "MulticastAddress"  : "224.0.0.1",
  "MulticastPort"     : "10001",
  "RTSPServer" : 0, "RtspServerComment" : "## Select RTSP Server > 0:Use RTSP Server from 'RTSPAddress:' 1:RPOS RTSP Server 2:V4L2 RTSP Server by mpromonet (auto selected if MulticastEnabled=true) 3:GStreamer RTSP Server",
  "PTZDriver" : "none", "PTZDriverComment": "## valid values are none,tenx,pelcod,visca and pan-tilt-hat",
  "PTZOutput" : "none", "PTZOutputComment": "## values are none (eg Tenx), serial and tcp",
  "PTZSerialPort" : "/dev/ttyUSB0",
  "PTZSerialPortSettings" : { "baudRate":2400, "dataBits":8, "parity":"none", "stopBits":1 },
  "PTZOutputURL": "127.0.0.1:9999",
  "PTZCameraAddress": 1,
  "DeviceInformation" : {
    "Manufacturer" : "RPOS Proxy",
    "Model" : "PROXY",
    "HardwareId" : ""
  },
  "logLevel" : 3, "logLevelComment": "## LogLevels are > 1:Error 2:Warning 3:Info 4:Debug",
  "logSoapCalls" : true
}
```

RogerHardiman, 2023-08-06:
> But it seems that it is instead returning the IP of the machine that is running RPOS instead of the one configured for the camera on my config for the RTSP stream, at least that's how I interpret [...]

Your config looks like RPOS will reply to GetStreamUri with "rtsp://admin:admin@192.168.1.251:554/1/h264major"

I notice this is quite different from the RPOS IP address which is 192.168.15.29.
Is there a typo, it your camera is at 192.168.15.251 and not .1.251 ?

If that is correct, can Android Onvifer see 192.168.1.xx and 192.168.15.xx.
Are they in the same Subnet? It is possible Onvifer is engaging NAT mode and thinks that the camera is outside the subnet range of RPOS and so replaces the IP Address that RPOS returns.

RogerHardiman, 2023-08-07:
So the fix for this is to run a RTSP Proxy on the RPOS machine.
Then RTSP connections go to the RPOS machine, which in turn reaches out to your actual RTSP camera.
I've used rtsp-simple-server and live555proxy for these sorts of things

philnagel, 2023-08-07:
Good idea, I will give that a shot. Thank you!

sravich1, 2023-12-26:
Hi ,
Below is my config file which I tried to configure to make raspberrypi4 ONVIF complaint...still not working
_{
  "NetworkAdapters" : ["awdl0","eth0", "wlan0", "en0"],
  "IpAddress" : "10.35.62.102",
  "ServicePort" : 8081,
  "Username" : "admin",
  "Password" : "admin",
  "CameraType" : "picam",
  "RTSPAddress" : "", "//":"Normally left blank. Used to set RTSP Server Address",
  "RTSPPort" : 8554,
  "RTSPName" : "h264",
  "MulticastEnabled"  : false,
  "RTSPMulticastName" : "h264m",
  "MulticastAddress"  : "224.0.0.1",
  "MulticastPort"     : "10001",
  "RTSPServer" : 0, "RtspServerComment" : "## Select RTSP Server > 1:RPOS RTSP Server 2:V4L2 RTSP Server by mpromonet (auto selected if MulticastEnabled=true)",
  "PTZDriver" : "none", "PTZDriverComment": "## valid values are none,tenx,pelcod,visca and pan-tilt-hat",
  "PTZOutput" : "none", "PTZOutputComment": "## values are none (eg Tenx), serial and tcp",
  "PTZSerialPort" : "/dev/ttyUSB0",
  "PTZSerialPortSettings" : { "baudRate":2400, "dataBits":8, "parity":"none", "stopBits":1 },
  "PTZOutputURL": "127.0.0.1:9999",
  "PTZCameraAddress": 1,
  "DeviceInformation" : {
    "Manufacturer" : "Raspberry Pi",
    "Model" : "4 B",
    "HardwareId" : ""
  },
  "logLevel" : 3, "logLevelComment": "## LogLevels are > 1:Error 2:Warning 3:Info 4:Debug",
  "logSoapCalls" : false
}_

I have changed the rtsp server to 0 ..getting rtspserver is not accessible
if I change to 3 rtspserver is working but getting some ffmeg issue..rtsp server runs but not able to get the video
_ffmpeg exec error: Error: Command failed: ffmpeg -fflags nobuffer -probesize 256 -rtsp_transport tcp -i rtsp://127.0.0.1:8554/h264 -vframes 1  -r 1 -s 640x360 -y /dev/shm/snapshot.jpg
ffmpeg version 4.3.6-0+deb11u1+rpt5 Copyright (c) 2000-2023 the FFmpeg developers
  built with gcc 10 (Debian 10.2.1-6)
  configuration: --prefix=/usr --extra-version=0+deb11u1+rpt5 --toolchain=hardened --incdir=/usr/include/aarch64-linux-gnu --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libdav1d --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librabbitmq --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sdl2 --disable-mmal --enable-neon --enable-v4l2-request --enable-libudev --enable-epoxy --enable-sand --libdir=/usr/lib/aarch64-linux-gnu --arch=arm64 --enable-pocketsphinx --enable-libdc1394 --enable-libdrm --enable-vout-drm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared
  libavutil      56. 51.100 / 56. 51.100
  libavcodec     58. 91.100 / 58. 91.100
  libavformat    58. 45.100 / 58. 45.100
  libavdevice    58. 10.100 / 58. 10.100
  libavfilter     7. 85.100 /  7. 85.100
  libavresample   4.  0.  0 /  4.  0.  0
  libswscale      5.  7.100 /  5.  7.100
  libswresample   3.  7.100 /  3.  7.100
  libpostproc    55.  7.100 / 55.  7.100
[tcp @ 0x55b474de60] Connection to tcp://127.0.0.1:8554?timeout=0 failed: Connection refused
rtsp://127.0.0.1:8554/h264: Connection refused

    at ChildProcess.exithandler (child_process.js:308:12)
    at ChildProcess.emit (events.js:314:20)
    at maybeClose (internal/child_process.js:1022:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:287:5) {
  killed: false,
  code: 1,
  signal: null,
  cmd: 'ffmpeg -fflags nobuffer -probesize 256 -rtsp_transport tcp -i rtsp://127.0.0.1:8554/h264 -vframes 1  -r 1 -s 640x360 -y /dev/shm/snapshot.jpg'
}_
similarly tried using sample_config_proxy.json...not gettting rtsp server working...
Kindly let me know where I should change to see rtsp working and then see the video .
Thanks
Ravi

sravich1, 2023-12-26:
look like rtsp is working for option 3 but ffmpeg decoder is failing ...thats the reason video is not seen when we try to connect..
any pointers here will help a lot.
Thanks
Ravi

sravich1, 2024-01-24:
Hi all,
trying to get the reason for FFMPEG error...i am using option 3 but not able to get rtsp going
**error**
ffmpeg exec error: Error: Command failed: ffmpeg -fflags nobuffer -probesize 256 -rtsp_transport tcp -i rtsp://127.0.0.1:8554/stream -vframes 1  -r 1 -s 640x360 -y /dev/shm/snapshot.jpg
ffmpeg version 4.3.6-0+deb11u1+rpt5 Copyright (c) 2000-2023 the FFmpeg developers

Kindly let me know how we can solve this....
--ravi