BreeeZe/rpos #118: Unable to get RTSP feed with USB Cam

EduardoRonchi, 2021-10-08:
Hello, first of all thanks for this project!

I am using a Raspberry Pi 4 with a RPI Cam V2 and a thermal USB Camera ([Lepton 3.5 + PureThermal Mini](https://groupgets.com/manufacturers/getlab/products/purethermal-mini-flir-lepton-smart-i-o-module)). 

For the RPI Cam V2 it works fine, but when I try to use the USB Camera I cannot retrieve the RTSP stream.

Everything seems fine and similar to the RPI Cam V2, but I cannot get the stream.

[image]

I am using a simple python script the read the stream

`
import cv2

vcap = cv2.VideoCapture("rtsp://termodrone:termodrone@192.168.137.230:8000/h264")
while(1):
    ret, frame = vcap.read()
    cv2.imshow('VIDEO', frame)
    cv2.waitKey(1)
`
The output of this program is the following:
`
[rtsp @ 0x226dfd0] method DESCRIBE failed: 503 Service Unavailable
Traceback (most recent call last):
  File "rtsp_viewer.py", line 6, in <module>
    cv2.imshow('VIDEO', frame)
cv2.error: OpenCV(4.4.0) /tmp/pip-wheel-2l8ccy47/opencv-python/opencv/modules/highgui/src/window.cpp:376: error: (-215:Assertion failed) size.width>0 && size.height>0 in function 'imshow'

`
It seems that the image is None. But the same config works fine when changing the device (from /dev/video0 to /dev/video2) from the usb cam to the RPI Cam.

My rposConfig.json file:
`
{
  "NetworkAdapters" : ["awdl0","eth0", "wlan0", "en0"], 
  "IpAddress" : "192.168.137.230",
  "ServicePort" : 10000,
  "Username" : "admin",
  "Password" : "admin",
  "CameraType" : "usbcam",
  "CameraDevice" : "/dev/video0",
  "RTSPAddress" : "", "//":"Normally left blank. Used to set RTSP Server Address",
  "RTSPPort" : 8000,
  "RTSPName" : "h264",
  "MulticastEnabled"  : false, "MulticastEnabledComment" : "## Multicast is not supported for USB camera",
  "RTSPMulticastName" : "h264m",
  "MulticastAddress"  : "224.0.0.1",
  "MulticastPort"     : "10001",
  "RTSPServer" : 3, "RtspServerComment" : "## Select RTSP Server > 1:RPOS RTSP Server 2:V4L2 RTSP Server by mpromonet (auto selected if MulticastEnabled=true)",
  "PTZDriver" : "none", "PTZDriverComment": "## valid values are none,tenx,pelcod,visca and pan-tilt-hat",
  "PTZOutput" : "none", "PTZOutputComment": "## values are none (eg Tenx), serial and tcp",
  "PTZSerialPort" : "/dev/ttyUSB0",
  "PTZSerialPortSettings" : { "baudRate":2400, "dataBits":8, "parity":"none", "stopBits":1 },
  "PTZOutputURL": "127.0.0.1:9999",
  "PTZCameraAddress": 1,
  "DeviceInformation" : {
    "Manufacturer" : "Raspberry Pi",
    "Model" : "4 B",
    "HardwareId" : ""
  },
  "logLevel" : 4, "logLevelComment": "## LogLevels are > 1:Error 2:Warning 3:Info 4:Debug",
  "logSoapCalls" : false
}
`

Any idea of what I am doing wrong? 
Note: The USB camera resolution is 160x120, it is a thermal camera. The camera is working, I can get the video by doing: `ffplay /dev/video0`

EduardoRonchi, 2021-10-15:
> just looked at the screen grab.
> It prints the gstreamer string on the screen and I can see it says
> "device=undefined" so it is not getting passed the right USB device path

Ok, thanks!
Ah, this printscreen is outdated. This “device=undefined” was due to the
facts that I was using the RPI config file. When I updated the config file
for the usbcam it shows the right device. I can’t post the image right now,
but I’ll update here after

Thanks!

EduardoRonchi, 2021-10-16:
Thats the updated output:
```
pi@raspberrypi:~/rpos $ node rpos.js
execSync('cat /proc/device-tree/model')
execSync('cat /proc/cpuinfo')
Read IP address from awdl0 failed
Read IP address from eth0 failed
Read IP address 192.168.137.230 from wlan0
Manufacturer : Raspberry Pi
Model : 4 B
HardwareId : 
SerialNumber : 10000000bc35458e
FirmwareVersion : 2.1.0
Starting camera settings webserver on http://192.168.137.230:10000/
execSync('v4l2-ctl --set-fmt-video=pixelformat=4')
execSync('v4l2-ctl --set-fmt-video=width=1280,height=720')
execSync('v4l2-ctl --set-parm=25')
execSync('v4l2-ctl --set-priority=3')
execSync('v4l2-ctl -l')
Controlvalue 'brightness' : 128
value: 128 not in range: 0 - 100
Controlvalue 'contrast' : 128
value: 128 not in range: -100 - 100
Could not retrieve Controlvalue 'saturation'
Could not retrieve Controlvalue 'red_balance'
Could not retrieve Controlvalue 'blue_balance'
Could not retrieve Controlvalue 'horizontal_flip'
Could not retrieve Controlvalue 'vertical_flip'
Could not retrieve Controlvalue 'power_line_frequency'
Could not retrieve Controlvalue 'sharpness'
Could not retrieve Controlvalue 'color_effects'
Could not retrieve Controlvalue 'rotate'
Could not retrieve Controlvalue 'color_effects_cbcr'
Could not retrieve Controlvalue 'video_bitrate_mode'
Could not retrieve Controlvalue 'video_bitrate'
Could not retrieve Controlvalue 'repeat_sequence_header'
Could not retrieve Controlvalue 'h264_i_frame_period'
Could not retrieve Controlvalue 'h264_level'
Could not retrieve Controlvalue 'h264_profile'
Could not retrieve Controlvalue 'auto_exposure'
Could not retrieve Controlvalue 'exposure_time_absolute'
Could not retrieve Controlvalue 'exposure_dynamic_framerate'
Could not retrieve Controlvalue 'auto_exposure_bias'
Could not retrieve Controlvalue 'white_balance_auto_preset'
Could not retrieve Controlvalue 'image_stabilization'
Could not retrieve Controlvalue 'iso_sensitivity'
Could not retrieve Controlvalue 'exposure_metering_mode'
Could not retrieve Controlvalue 'scene_mode'
Could not retrieve Controlvalue 'compression_quality'
Binding DeviceService to http://192.168.137.230:10000/onvif/device_service
Binding MediaService to http://192.168.137.230:10000/onvif/media_service
Binding PTZService to http://192.168.137.230:10000/onvif/ptz_service
Binding ImagingService to http://192.168.137.230:10000/onvif/imaging_service
discovery_service started
Starting Live555 rtsp server
spawn('./python/gst-rtsp-launch.sh', [-P,8000,-u,h264,-W,1280,-H,720,-d,/dev/video0], undefined)
media_service started
imaging_service started
device_service started
ptz_service started
rtspServer: -P 8000 -u h264 -W 1280 -H 720 -d /dev/video0

rtspServer: INFO:root:StreamServer initialized

rtspServer: INFO:root:Video settings loaded from v4l2ctl.json

rtspServer: DEBUG:root:StreamServer.launch
INFO:root:USB camera ignored most of the parameters

rtspServer: DEBUG:root:( v4l2src device=/dev/video0 brightness=50 contrast=0 saturation=0 ! image/jpeg,width=1280,height=720,framerate=30/1 ! jpegdec ! clockoverlay ! omxh264enc target-bitrate=10000000 control-rate=variable ! video/x-h264,profile=baseline ! h264parse ! rtph264pay name=pay0 pt=96 )
INFO:root:Starting service on port 8000 at url /h264

rtspServer: INFO:root:Running RTSP Server
```

I also enabled verbose and after some time (under one minute) I get this infos:
```
Discovery received from 192.168.137.1
(node:359) [DEP0005] DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.
DeviceService received request GetSystemDateAndTime
DeviceService received request GetCapabilities
MediaService received request GetProfiles
MediaService received request GetStreamUri
MediaService received request GetSnapshotUri
PTZService received request GetConfigurations
PTZService received request GetConfigurationOptions
```
Folloing are the print screen of the output
[image]
[image]

GeniusOnline, 2021-11-24:
@RogerHardiman @EduardoRonchi any update on this?
i'm having the same issue using a v4l2loopback device #123