BreeeZe/rpos #121: Multi-channel / Multi-RTSP streaming

Shell-222, 2021-11-04:
Hello all,

I am using RPOS as a video stream proxy which takes in RTSP stream and output it as a onvif device to the network.

At the moment, I succeeded with one RTSP stream. However, my application requires me to stream multiple channels in one RPOS instance. Say, I have three cameras, each of which has a RTSP stream. What/How should I configure to make the js node to take in all three RTSP url and appear in the ONVIF Device Manager as one device with three video channels? 

I attached a screenshot of what I am trying to achieve. Hope that helps explaining what I want to do.

Thanks in advance! 
[image]

RogerHardiman, 2021-11-04:
I understand what you are trying to do.

Unfortunately with RPOS this is not possible with config file settings as RPOS was written to tell ONVIF users that it has only 1 camera

It will require code changes to RPOS to the ONVIF messages that tell viewers (like ODM) how many Video Sources there are, and then needs code changes to have additional Profiles for each video Source and needs changes to the GetJPEGSnapshot and GetStreamUri functions to return the correct addresses from a list in the config file.

It is about 3 days of programming work to write and test this which is more than I can give in my spare time.

Roger

Shell-222, 2021-11-04:
> It is about 3 days of programming work to write and test this which is more than I can give in my spare time.

Thanks for replying. May I please have a quote if you were to make it happen?

RogerHardiman, 2021-11-04:
send me an email to 
[image][image]

I can make RPOS proxy as many RTSP streams as you want via the ONVIF XML messages but in the real world CCTV software expects a maximum of 16 streams per ONVIF Device (eg a 16 channel Bosch or Pelco analogue encoder)
So you may have to run a few copies of RPOS on different ports if needed.

cdaher78, 2021-11-25:
I'm very interested on this development. 

In general on IP CCTV world, basic camears has 2 independent video streams, usually for vewing (low resolution, high framerate) and recording (high resolution, low framerate) givin' us great flexibility on VMS settings. Maybe adding another ONVIF stream profile, and let us set Resolution, Framerate, Bitrate Target, Bitrate Mode (CBR, VBR) parameters for both stream profiles will put this project on a stellar level.

mrdown11, 2022-01-25:
Hi, I'm very interested in this functionality as well. I am using rpos on a stereoPi (with 2 cameras). Right now I've made a trigger to restart the rtsp server for the second camera, but this has some issues with reconnecting sometimes, and it does not allow simultanious streaming. I'm also worried for high resolutions the pi CM3 is not powerful enough for multiple streams, but the stereopi v2 with CM4 is on the way.

VCTGomes, 2023-08-01:
> send me an email to [image][image]
>
> I can make RPOS proxy as many RTSP streams as you want via the ONVIF XML messages but in the real world CCTV software expects a maximum of 16 streams per ONVIF [...]

It'd be really cool to be implemented and having an extra streaming with low resolutions. My NVRs uses it to provide multiple cameras view:
[image]
On this example, I'm on multi camera viewer, so my NVR chooses to use low resolution streaming, how RPOS doesn't delivers it, the image is black.

Same happens on my NVR app, where it delivers low res to save mobile data