BreeeZe/rpos #122: Baudrate is not stable

bagrintech, 2021-11-14:
I set the port speed to 4800 baudrate.
When I send the zoom in command, the baudrate is 4800 5 times out of 10. But 5 times out of 10 the baudrate is 2400 randomly. I observe this with an oscilloscope. An interesting fact is that this happened after the last upgrade of RPI 3. Everything works fine on RPI 4.

RogerHardiman, 2021-11-14:
I think the Pi 3 has two UARTs. One uses proper UART hardware inside the Soc
The other uses a 'software UART'. I think on the Pi3 the real UART got wired to the Bluetooth chip and the GPIO pins were mapped to the software UART, which did have bugs when CPU speeds where changed, overclocked or throttled. But those bugs were 3 years ago or something like that.
There were instructions to switch the GPIO pins over to the real UART.

So this may the cause of the problem.

RogerHardiman, 2021-11-15:
https://openenergymonitor.org/forum-archive/node/12311.html

this page had some stuff on re-mapping the UART from the Bluetooth chip to the pins.
I'd forgotten all about this problem as I use USB to Serial Adapters (usually USB to RS485 adapters) on the Pi for Serial comms

closed 2022-03-12 by RogerHardiman