BreeeZe/rpos #125: rpos.js as systemd service

cdaher78, 2021-11-25:
Hello brothers.

If I run node rpos.js directly at /home/pi/rpos the service inits correctl without any issues.

But I need to make it run automatically and tried to make rpos.js to init as a service on raspberry reboot. 

When I start the service it soon fails.

``service parameters on /lib/systemd/system/onvifcam.service``

[Unit]
Description=ONVIF/RTSP Server
After=network.target

[Service]
Type=simple
User=pi
ExecStart=/usr/bin/node /home/pi/rpos/rpos.js
#Restart=on-failure

[Install]
WantedBy=multi-user.target

``systemctl status onvifcam.service``

● onvifcam.service - ONVIF/RTSP Server
   Loaded: loaded (/lib/systemd/system/onvifcam.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Thu 2021-11-25 13:01:01 -03; 15s ago
  Process: 1091 ExecStart=/usr/bin/node /home/pi/rpos/rpos.js (code=exited, status=1/FAILURE)
 Main PID: 1091 (code=exited, status=1/FAILURE)

Nov 25 13:01:01 raspberrypi-dev node[1091]:     at Object.<anonymous> (/home/pi/rpos/rpos.js:41:15)
Nov 25 13:01:01 raspberrypi-dev node[1091]:     at Module._compile (internal/modules/cjs/loader.js:778:30)
Nov 25 13:01:01 raspberrypi-dev node[1091]:     at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
Nov 25 13:01:01 raspberrypi-dev node[1091]:     at Module.load (internal/modules/cjs/loader.js:653:32)
Nov 25 13:01:01 raspberrypi-dev node[1091]:     at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
Nov 25 13:01:01 raspberrypi-dev node[1091]:     at Function.Module._load (internal/modules/cjs/loader.js:585:3)
Nov 25 13:01:01 raspberrypi-dev node[1091]:     at Function.Module.runMain (internal/modules/cjs/loader.js:831:12)
Nov 25 13:01:01 raspberrypi-dev node[1091]:     at startup (internal/bootstrap/node.js:283:19)
Nov 25 13:01:01 raspberrypi-dev systemd[1]: onvifcam.service: Main process exited, code=exited, status=1/FAILURE
Nov 25 13:01:01 raspberrypi-dev systemd[1]: onvifcam.service: Failed with result 'exit-code'.

``cat /var/log/daemon.log``

Nov 25 13:00:57 raspberrypi-dev systemd[1]: Started ONVIF/RTSP Server.
Nov 25 13:01:01 raspberrypi-dev node[1091]: fs.js:114
Nov 25 13:01:01 raspberrypi-dev node[1091]:     throw err;
Nov 25 13:01:01 raspberrypi-dev node[1091]:     ^
Nov 25 13:01:01 raspberrypi-dev node[1091]: Error: ENOENT: no such file or directory, open './rposConfig.json'
Nov 25 13:01:01 raspberrypi-dev node[1091]:     at Object.openSync (fs.js:443:3)
Nov 25 13:01:01 raspberrypi-dev node[1091]:     at Object.readFileSync (fs.js:343:35)
Nov 25 13:01:01 raspberrypi-dev node[1091]:     at Object.<anonymous> (/home/pi/rpos/rpos.js:41:15)
Nov 25 13:01:01 raspberrypi-dev node[1091]:     at Module._compile (internal/modules/cjs/loader.js:778:30)
Nov 25 13:01:01 raspberrypi-dev node[1091]:     at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
Nov 25 13:01:01 raspberrypi-dev node[1091]:     at Module.load (internal/modules/cjs/loader.js:653:32)
Nov 25 13:01:01 raspberrypi-dev node[1091]:     at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
Nov 25 13:01:01 raspberrypi-dev node[1091]:     at Function.Module._load (internal/modules/cjs/loader.js:585:3)
Nov 25 13:01:01 raspberrypi-dev node[1091]:     at Function.Module.runMain (internal/modules/cjs/loader.js:831:12)
Nov 25 13:01:01 raspberrypi-dev node[1091]:     at startup (internal/bootstrap/node.js:283:19)
Nov 25 13:01:01 raspberrypi-dev systemd[1]: onvifcam.service: Main process exited, code=exited, status=1/FAILURE
Nov 25 13:01:01 raspberrypi-dev systemd[1]: onvifcam.service: Failed with result 'exit-code'.

I have added /home/pi/rpos to PATH.

node version is ``v8.17.0``
npm version is ``6.13.4``
OS is  ``Raspberry Pi OS 10 (Buster)``

What am I missing here? I'm not aware of Node.js particularities.

Thanks.

RogerHardiman, 2021-11-27:
The error log says
   Error: ENOENT: no such file or directory, open './rposConfig.json'

It cannot find the config file

You probably need to set the folder it runs from in the Systemd config file.
It will be called something like Run Directory, or PWD or CWD or Start Folder

cdaher78, 2021-11-27:
Hey Roger,

Thanks for your comment.

After some research and struggling I've made it work. At the service file was needed to point for a "WorkingDirectory"

There goes working service config:

[Unit]
Description=ONVIF/RTSP Server
Requires=network.target
After=network.target

[Service]
Type=simple
Restart=always
StandardOutput=syslog
StandardError=syslog
SyslogIdentifier=onvifcam
User=pi
WorkingDirectory=/home/pi/rpos
ExecStart=/usr/bin/node rpos.js

[Install]
WantedBy=multi-user.target

It would be great if this project were compatible with Bullseye and newer versions of Node.js. and multi streaming capability (at least 2 different and editable streams with his own resolution, framerate, bitrate).

Congrats to you and the contributors for this awesome development job.

RogerHardiman, 2022-03-12:
Hi
I've added your startup service to the project. Many thanks
Roger

closed 2022-03-12 by RogerHardiman