BreeeZe/rpos #126: Integrating RPI camera running ONVIF with genetec security center

PythonMoore, 2021-11-26:
Hello team,

I am trying to connect my  Raspberry PI Zero W camera running RPOS 2.0.6 with Genetec's Security Center. I am able to connect to the camera using Onvif Device manager and view the feed there after logging in. 

When I am trying to connect with the Genetec Security Center, the software is able to discover the camera but unable to connect.

Any help?

[image]

RogerHardiman, 2021-11-26:
When you add the camera, make sure you select ONVIF and BASIC mode (not Full Mode)

Genetec use some parts of the ONVIF spec - (creation of Profile Tokens) that RPOS has not implemented.
Adding in BASIC mode tells Genetec not to expect all ONVIF commands to work

PythonMoore, 2021-11-26:
Hello @RogerHardiman
Yes, I am using Onvif Basic mode itself for searching the Camera. Only after doing that did it start discovering the camera. But I am unable to connect with it.

RogerHardiman, 2021-11-26:
which version of genetec are you using.

PythonMoore, 2021-11-26:
Genetec Security Center (GSC) Base Package - Version 5.10

RogerHardiman, 2021-11-27:
I'll check what version i have.
But I just noticed your first post said RPOS 2.0.6

the package.json file reminded me we want to 2.1.0 about 6 months ago and I've made some changes since then but not bumped the version number.
Are you getting the source code directly from github, or from some other packaged up installation?

PythonMoore, 2021-11-27:
I took the package from the tagged one which was out in May. I was trying with the latest version but I guess it doesn't support v4l2rtspserver anymore and uses gstreamer instead. My stream quality doesn't come well on gstreamer.

RogerHardiman, 2021-11-27:
There have been lots of changes since 2.0.6

The current code in the master branch on github is the place for the latest version.

RPOS still supports v4l2rtspserver.
You can pick, in the config file, if you want v4l2rtsp server or if you want to use gstreamer.

Gstreamer is there for video sources that do not generate H264 video themselves, like normal USB cameras.
Gstreamer is also there for when we want to overlay text onto the video stream, where it can grab any video source (USB or Pi cameras, overlay date and time, and then encode and then send out.

But v4l2rtspserver is still there.

Roger

PythonMoore, 2021-11-28:
Ok @RogerHardiman. I will try it and let you know the result. Please do let me know if the version works well for you with Genetec

RogerHardiman, 2022-03-12:
Did you get RPOS working with Genetec 5.10 ?

DarwinAnim8or, 2023-04-28:
> 

Sorry for replying to an old thread, but I'm currently setting up rpos as a proxy server for a stream, and Genetec 5.11 does work if Onvif is set to "basic" like you said. That being said, I am having an issue with getting Genetec to actually stream the video stream correctly. 

The "ONVIF Device Manager" correctly sees the rpos instance + can view the RTSP stream, but Genetec is incapable of playing back the proxied RTSP stream. Adding the RTSP stream manually (not proxied) works. 

I am running Windows currently, the server I'm using is based off the MediaLAN RTSP code you may be familiar with. I am seeing logs of Genetec attempting to connect but it doesn't play for some reason (loses connection)

Do you have any advice by any chance? 
In either case, thank you for reading and thank you for all the work on RPOS!

RogerHardiman, 2023-04-28:
Hi,
Don't know the MediaLAN RTSP Server.
The Genetec Basic vrs Full ONVIF code is because RPOS does not allow the creation of custom ONVIF Profiles and the Full mode creates them and checks they work.
For the video, I recall Genetec was fussy on some NAL payload formats and numbers of chunks when working on another RTSP stream, but if you said it worked adding as a RTSP source that is probably not the problem.
Is MediaLAN running on the same device as RPOS? Some VMS systems don't like it when the RTSP URL points at another IP address.
Beyond that I'm not sure I can help. My Genetec trial licence expired so I cannot run it any more.

DarwinAnim8or, 2023-04-28:
> Hi, Don't know the MediaLAN RTSP Server. The Genetec Basic vrs Full ONVIF code is because RPOS does not allow the creation of custom ONVIF Profiles and the Full mode creates them and checks they [...]

Hi! Thanks for the speedy reply; The current setup is as follows: 
Machine 1:
  -RTSP Server running on port 8554
  -RPOS set up as proxy for the above

Machine 2:
  -Genetec installation 

I'm not sure where the issue may be, unless if RPOS also wants port 8554 when running as proxy? 
My eventual goal for this all is to be able to intercept the PTZ-commands by writing a custom "driver" for RPOS