BreeeZe/rpos #130: Stream ONVIF to be accepted by NVR

rtek1000, 2022-01-18:
Hello,

Since you already know how to implement the main part of the Onvif protocol,

Would it be very difficult to add the functionality so that an NVR that works with the Xmeye App can find and work with the streaming provided by the Raspberry Pi?

I believe that it is not so easy to implement the Onvif protocol in ESP32, due to limited RAM memory, 

But it would be very useful to be able to count on the conversion of the stream generated by ESP32-CAM so that the NVR can make it available over the internet via P2P in the Xmeye App or similar.

It's really not the best solution, but it might be safer than leaving the ESP32-CAM's IP address open on the router and accessible on the internet.

I found another thread about this, but it doesn't seem to have any success yet:

https://github.com/bkeevil/esp32-cam/issues/5

RogerHardiman, 2022-01-18:
I looked at the other thread. if you used the RTSP stream firmware then what you'd have to do in RPOS is use RPOS Proxy Mode where the NVR talks to RPOS but when the NVR asks RPOS for the URL of the RTSP stream, RPOS returns a string that is the URL of the ESP32 camera.

It may work.
But there are some issues I can see already.
When an NVR talks to an ONVIF device, it asks what format the video stream is, and RPOS says "It's H264" whereas the real RTSP stream will be MJPEG. So that needs a change to RPOS if the NVR complains.
Also some NVRs assume the RTSP source is the same IP as the ONVIF service and complain when RPOS returns an IP address that is he ESP32 IP address

But.... it may just work fine.

You cam try it and see and then we can have a go at fixing the issues but I have to do this in spare time so it could be several months before I get chance to work on it (if at all)

Tempting to buy an ESP32 cameras myself just to play with.

Finally... many NVRs just let you add a camera manually and let you select ONVIF or RTSP streams so you may just be able to add it manually.

RogerHardiman, 2022-01-18:
just out of interest, which version of the ESP-32 module do you use.
I've never owned one before but see there are WROOM and WROVER versions and don't know which is better, as I'll probably buy one to play with

rtek1000, 2022-01-18:
Hi,

Thanks for the explanation, the NVR (actually a hybrid DVR in my case) also accepts another type of camera other than ONVIF, if I remember correctly it's like a NET camera, but I haven't found anything about this protocol yet, unfortunately there's no way to put it many configurations,

These chinese devices are usually made on demand and then cloned, so actually those who are just cloning may not even know how to improve this, besides possibly being an older architecture, and even then they put it up for sale to use the parts stock .

About ESP32 with Camera:

You can search it by "ESP32-CAM"

You can see more specs here:

https://www.arducam.com/esp32-machine-vision-learning-guide/

https://www.hackster.io/KDPA/esp32-cam-video-surveillance-smart-camera-7f9a63

rtek1000, 2022-01-18:
There is a possibility to make any IP camera or NVR (or DVR) available just by opening ports on the router, and if the Internet connection does not have a fixed IP, a DDNS service can also be used, there are several models of routers that already have this settings.

https://www.securitycameraking.com/securityinfo/how-to-connect-to-your-dvr-over-the-internet/

One such router is the TL-MR3020, which can access the Internet via ETHERNET port (RJ45 connector) and 3G/4G (USB port for external modem), in addition to enabling automatic switching if the main Internet connection fails.

https://www.tp-link.com/en/support/faq/1491/

But about the ESP32-CAM service I don't know if it supports login to have a minimum of security to be used in this way.

I found this option:

> Using ngrok tunnel service
> After signing up for the account, go to your email address and activate your account by clicking on the activation link.
> Now login to your account and click on the ‘Auth’ button’ to get your Tunnel Authtoken as shown in the figure below. Copy this Tunnel Authtoken, we will need it later on.

https://www.elementzonline.com/blog/Accessing-ESP32-CAM-Video-Streaming-from-anywhere-in-the-world

RogerHardiman, 2022-01-18:
Thanks for the link.
I may buy one and have a play, but unable to promise any timescale as have loads on at the moment so little spare time.

But in the mean time, we can make it work wit these changes if you are up for some testing

1) There is a Python folder in RPOS. This contains a simple Python program that gets images from a USB camera OR the Pi's camera and feeds it into a H264 encoder and a built in RTSP server using GStreamer.
We will modify this Python so that it asks GStreamer to get the images from the ESP32-CAM and pass them into the H264 encoder and RTSP Server.
2) In the RPOS config you need to select Gstreamer RTSP Server mode

And that should be it.
(there is a RPOS Proxy mode, but forget that for now)

This should be enough to make your RPOS device pull in video from the ESP32-CAM and re-distribute it as H264 video to your Hybrid DVR.

rtek1000, 2022-01-18:
I think it's interesting to do these tests, but unfortunately I also have loads to solve, so I've been talking to people who have more knowledge and maybe reach a solution more easily, than me learning to make the changes on my own. Anyway, I appreciate your attention.

I will try to test this as soon as possible, I have Raspberry Pi 1 and 3.

My ESP32-CAM are still coming.

sbcdave, 2024-09-30:
> Thanks for the link. I may buy one and have a play, but unable to promise any timescale as have loads on at the moment so little spare time.
>
> But in the mean time, we can make it work wit these [...]

I'm going to try this now, thanks for the tip

sbcdave, 2024-09-30:
I couldn't figure out what to modify in gst-rtsp-launch.py so I set RTSPServer to 3 and CameraType to Other, but I'm having the same issue, i.e. it seems to connect, it gets info from the camera on framerate and resolutions, and the name, but it stalls at the GetURI call on the MediaService and never kick off a soap for the ImagingService