BreeeZe/rpos #134: Unable to install rpos

jonathan84clark, 2022-04-12:
Hello,

I am attempting to install rpos on a Raspberry PI3 compute module. The OS on this module was updated to Buster and had a previous install of rpos that was working on Jesse.

**Steps to Reproduce**
1. sudo apt-get update
2. sudo apt-get upgrade
3. sudo apt install nodejs npm
4. sudo npm install -g n
5. sudo n install 12
6. git clone https://github.com/BreeeZe/rpos.git
7. cd rpos
8. npm install
Here is the output I am seeing:
npm ERR! Error while executing:
npm ERR! /usr/bin/git ls-remote -h -t git://github.com/BreeeZe/node-soap.git
npm ERR!
npm ERR! fatal: remote error:
npm ERR!   The unauthenticated git protocol on port 9418 is no longer supported.
npm ERR! Please see https://github.blog/2021-09-01-improving-git-protocol-security-github/ for more information.
npm ERR!
npm ERR! exited with error code: 128
npm WARN tar ENOENT: no such file or directory, lstat '/home/pi/rpos/node_modules/.staging/typescript-0a2c1eea/lib/it'

Any help on this would be great!

Thanks,

Jonathan L Clark

jonathan84clark, 2022-04-12:
Just checked a few other things. It looks like the BreeeZe node-soap fork is broken. It no longer complies with github' standards as of 1/2022. If you go into the rpos directory and do a
npm install soap
It will fail.
So this is a relatively new issue. I suspect that it can be resolved by modifying either the git or npm settings. It looks like the node-soap repository that BreeeZe was forked from still works and there was a recent commit that appears to have addressed this issue. https://github.com/vpulim/node-soap/commit/94fada0b0b56059c774bc312f662349f8652e6d2
But I'm getting a little out of my depth here.

BreeeZe, 2022-04-13:
Looks like the git:// protocol is no longer supported, I changed it to https:// and created a pull request (#136) for @RogerHardiman to look at

osvikvi, 2022-10-29:
HEllo, I too had some issues while installing this newer version. I already have a 3b+ and two zero-s running RPOS. Since a couple of days I-ve been trying to set/up on a zero2. and have it finally sort/of running. I-ll describe my steps to success below>

Some of the steps in STEP 1 I actually did during the entire process but I guess they could just as well be in step 1.
Out of the box lite install (A port of debian Bullseye without desktop - autoboot to console)

STEP 1
	$ sudo apt-get update
	$ sudo apt-get upgrade
 	$ sudo nano /etc/apt/sources.list
	          add local mirror: http://raspbian.mirror.garr.it/mirrors/raspbian/raspbian/
	$ sudo apt-get install libtool
	$ sudo apt-get install git
	$ sudo apt --fix-broken install
		now libuv1-dev should be installed
STEP 2
	$ sudo apt install nodejs npm
	$ sudo npm install -g n
	$ sudo n install 12	
		*LOGOUT/LOGIN
	$ node -v
		v12.22.12
	$ npm -v
		6.14.16
STEP 3
	$ git clone https://github.com/BreeeZe/rpos.git
	$ CD RPOS
	$ npm install
STEP 4
	$ npx gulp
STEP 5
	$ sudo apt install git gstreamer1.0-plugins-base
	$ sudo apt install git gstreamer1.0-plugins-bad
	$ sudo apt install git gstreamer1.0-plugins-good
	$ sudo apt install git gstreamer1.0-plugins-ugly
	$ sudo apt install git gstreamer1.0-tools
	$ sudo apt install git libgstreamer1.0-dev
	$ sudo apt install git libgstreamer1.0-0-dbg
	$ sudo apt install git libgstreamer1.0-0
	$ sudo apt install git libgstrtspserver-1.0.0
	$ sudo apt install git libgstreamer-plugins-base1.0-dev
	$ sudo apt install git gtk-doc-tools
	$ sudo apt install git gstreamer1.0-omx-rpi
	$ sudo apt install git gstreamer1.0-omx
  		some of these give an error...
	$ sudo apt-get install python3-gi gir1.2-gst-plugins-base-1.0 gir1.2-gst-rtsp-server-1.0
  		here I  had to specify python3
STEP 5.c.1b ~*
	$cd ..
	$  git clone https://github.com/thaytan/gst-rpicamsrc.git
	$ cd gst-rpicamsrc
	$ sudo apt-get install autoconf
	$ ./autogen.sh
	$ make
	$ sudo make install
	$ cd ..
	$ gst-inspect-1.0 rpicamsrc

This is how I could make RPOS work on my zero2. however, I must add that its still not running smooth, my older pi zero-s are both capable of 1080 x 1920 10fps without issues, but on this zero2 which is significantly more powerful even overclocked to 1300Mhz, 1280x720 2fps is already difficult. @BreeeZe @RogerHardiman would you have any idea why this lag?

RogerHardiman, 2022-10-29:
I don't own a Pi Zero or a Zero 2 and never tried RPOS on these devices.

I would think the best RTSP stream is probably using mpromonet's RTSP server and not Gstreamer on the Zero / Zero 2

osvikvi, 2022-10-30:
thanks a lot, I've never tried mpromonet actually I understood Gstreamer was suggested for use with usbcam. I'm installing it now....

Indeed, the pi reports: "Only Gstreamer RTSP is supported for USB camera video"

RogerHardiman, 2022-10-30:
oh, I assumed you were using the Pi's camera.
the mpromonet RTSP server needs a H264 v4l2 source, eg the Pi camera and hardware encoding, or one of the USB cameras that encodes to h264 in the camera (most return a YUV image or JPEG image over the USB bus).
Although mpromonet may be able to use the JPEG streams now.

Now I know you are on a USB camera I expect the issue is the CPU time taken to receive the USB images and re-encode as H264.
It may be worth seeing if you can stream JPEG from the USB camera and get gstreamer to stream the JPEG images over RTSP.

osvikvi, 2022-10-30:
Thanks for the suggestion, I'll try to figure it out.

Actually I'm using a 10€ UVC - HDMI->USB in combination with Tandberg / Cisco VISCA PTZ camera's.

osvikvi, 2022-10-31:
that was my idea as well, I have a whole bunch of those Tandberg TTC08 camera's and wanted to benefit the great optics and PTZ functionality. but only one is working on my Raspi3b+ running Jessie, but can't seem to get a proper videostream on my zero2 running bullseye.

do you have any luck getting a stable image from this HDMI source?

RogerHardiman, 2022-11-01:
I use a USB HDMI adapter on a Pi4 (actually a Pi-400 running headless) which is on an older Jessie OS release (or maybe the Bullseye legacy release)

The HDMI image was fine. I just need to capture at a lower resolution (set via v4l2ctl I think) as grabbing at full resolution overloads the USB data bus and eats too much CPU reading the raw image in.

osvikvi, 2022-11-01:
Success, I could manage to get a decent image STREAMING 1080p @2fps HDMI over usb
In rposConfig.json set
	  "RTSPServer" : 2, 
In camera .js,:
	  disable the line that prevents from using MPromonet server ('yxzusbcam')
	    (this.config.CameraType == 'usbcam') 
	  set resolution and framerate to 1920 x 1080 

start the server and go!

it's not really stable, i do loose the signal from time to time...
..trying to figure out how to control v4l2...

RogerHardiman, 2022-11-01:
that's great news

osvikvi, 2022-11-01:
Thanks, I think so too ;D... 
It's all in camera.js I could enable verbose logging on the bottom and apparently the server still pushes 25fps while higher up, I requested only 2fps. setting -F 2 again helps. Now the stream is stable. 

                this.rtspServer = utils.spawn("v4l2rtspserver", ["-v", "-P", this.config.RTSPPort.toString(), "-u", this.config.RTSPName.toString(),"-F", "2", "-W", this.settings.resolution.Width.toString(), "-H", this.settings.resolution.Height.toString(), "-t", this.config.CameraType, "-d", (this.config.CameraDevice == "" ? "auto" : this.config.CameraDevice)]);