BreeeZe/rpos #143: RTSP Server fails when using virtual USB camera

noomSik, 2022-11-02:
Hello. RPOS works wonderfully when using the integrated webcam, but I'm trying to use a virtual camera to be able to pass any kind of feed like video from a USB capture device or from an incoming rtsp stream, through rpos (and therefore ONVIF).

The only error I get is the FFMPEG one:

`ffmpeg exec error: Error: Command failed: ffmpeg -fflags nobuffer -probesize 256 -rtsp_transport tcp -i rtsp://127.0.0.1:8554/h264 -vframes 1  -r 1 -s 640x360 -y /dev/shm/snapshot.jpg
ffmpeg version 4.2.7-0ubuntu0.1 Copyright (c) 2000-2022 the FFmpeg developers
  built with gcc 9 (Ubuntu 9.4.0-1ubuntu1~20.04.1)
  configuration: --prefix=/usr --extra-version=0ubuntu0.1 --toolchain=hardened --libdir=/usr/lib/x86_64-linux-gnu --incdir=/usr/include/x86_64-linux-gnu --arch=amd64 --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sdl2 --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-nvenc --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared
  libavutil      56. 31.100 / 56. 31.100
  libavcodec     58. 54.100 / 58. 54.100
  libavformat    58. 29.100 / 58. 29.100
  libavdevice    58.  8.100 / 58.  8.100
  libavfilter     7. 57.100 /  7. 57.100
  libavresample   4.  0.  0 /  4.  0.  0
  libswscale      5.  5.100 /  5.  5.100
  libswresample   3.  5.100 /  3.  5.100
  libpostproc    55.  5.100 / 55.  5.100
[rtsp @ 0x55feb67b0340] method DESCRIBE failed: 503 Service Unavailable
rtsp://127.0.0.1:8554/h264: Server returned 5XX Server Error reply

    at ChildProcess.exithandler (node:child_process:400:12)
    at ChildProcess.emit (node:events:513:28)
    at maybeClose (node:internal/child_process:1093:16)
    at Process.ChildProcess._handle.onexit (node:internal/child_process:302:5) {
  code: 1,
  killed: false,
  signal: null,
  cmd: 'ffmpeg -fflags nobuffer -probesize 256 -rtsp_transport tcp -i rtsp://127.0.0.1:8554/h264 -vframes 1  -r 1 -s 640x360 -y /dev/shm/snapshot.jpg'
}`

which i assume has nothing to do with the problem and just means that the RTSP server failed to start.

I checked everywhere and resolutions are all set to the same. I tried using V4L2Loopback and OBS to create the virtual cam, both fail.

The onvif tester I use reports that connection is succesful but video streaming failed. The FFMPEG error is thrown out when trying to connect to the ONVIF stream.

I understand V4L2loopback generates a rawvideo feed because ffmpeg does only support outputting rawvideo to v4l2, but as far as I know, RPOS does no longer need an h264 input right? Also, shouldn't OBS be outputting h264 video already?

I've read something about using VLC's built in rtsp server but i've got no idea on how to start on that so any help is appreciated. 

I know this is somewhat chaotic but I'm really lost on this and I need to finish it. Thanks in advance

RogerHardiman, 2022-11-04:
The ffmpeg issue you see is just a trick (a hack) to get a JPEG image for the ONVIF GetSnapshotURI command.
I needed an image from the camera to make a JPEG, so the code tries to make a RTSP connection to itself, get 1 frame of video, and use that as the JPEG image.

So what you are seeing is just an error that says the JPEG snapshot code could not connect to 127.0.0.1

That tells me the main RTSP server is not running.

noomSik, 2022-11-08:
> The ffmpeg issue you see is just a trick (a hack) to get a JPEG image for the ONVIF GetSnapshotURI command. I needed an image from the camera to make a JPEG, so the code tries to make a RTSP [...]

The only way I managed to make it work is encoding my video feed in MJPEG. Does RPOS not support H264 video from webcams?