BreeeZe/rpos #151: Not getting any rtsp

enboig, 2023-01-13:
After installation, I can open the html interface to configure the camera, but cannot get any stream using vlc at `rtsp://admin:******@192.168.1.111:8554`
My log shows some errors:
```
pi@camera:~/rpos $ node rpos.js
Read IP address 192.168.1.111 from wlan0
Manufacturer : Raspberry Pi
Model : 3 B
HardwareId : 
SerialNumber : 00000000371183ed
FirmwareVersion : 2.1.0
Starting camera settings webserver on http://192.168.1.111:8081/
The pixelformat index was invalid
Could not retrieve Controlvalue 'red_balance'
Could not retrieve Controlvalue 'blue_balance'
Could not retrieve Controlvalue 'horizontal_flip'
Could not retrieve Controlvalue 'vertical_flip'
Could not retrieve Controlvalue 'power_line_frequency'
Could not retrieve Controlvalue 'sharpness'
Could not retrieve Controlvalue 'color_effects'
Could not retrieve Controlvalue 'rotate'
Could not retrieve Controlvalue 'color_effects_cbcr'
Could not retrieve Controlvalue 'video_bitrate_mode'
Could not retrieve Controlvalue 'video_bitrate'
Could not retrieve Controlvalue 'repeat_sequence_header'
Could not retrieve Controlvalue 'h264_i_frame_period'
Could not retrieve Controlvalue 'h264_level'
Could not retrieve Controlvalue 'h264_profile'
Could not retrieve Controlvalue 'auto_exposure'
Could not retrieve Controlvalue 'exposure_time_absolute'
Could not retrieve Controlvalue 'exposure_dynamic_framerate'
Could not retrieve Controlvalue 'auto_exposure_bias'
Could not retrieve Controlvalue 'white_balance_auto_preset'
Could not retrieve Controlvalue 'image_stabilization'
Could not retrieve Controlvalue 'iso_sensitivity'
Could not retrieve Controlvalue 'exposure_metering_mode'
Could not retrieve Controlvalue 'scene_mode'
Could not retrieve Controlvalue 'compression_quality'
Binding DeviceService to http://192.168.1.111:8081/onvif/device_service
Binding MediaService to http://192.168.1.111:8081/onvif/media_service
Binding PTZService to http://192.168.1.111:8081/onvif/ptz_service
Binding ImagingService to http://192.168.1.111:8081/onvif/imaging_service
discovery_service started
Starting rtsp server
media_service started
ptz_service started
imaging_service started
device_service started
rtspServer: /home/pi/rpos/./python/gst-rtsp-launch.py:60: PyGIDeprecationWarning: GObject.MainLoop is deprecated; use GLib.MainLoop instead
  self.mainloop = GObject.MainLoop()

rtspServer: /home/pi/rpos/./python/gst-rtsp-launch.py:78: PyGIDeprecationWarning: Since version 3.11, calling threads_init is no longer needed. See: https://wiki.gnome.org/PyGObject/Threading
  GObject.threads_init()

rtspServer: INFO:root:StreamServer initialized

rtspServer: INFO:root:Video settings loaded from v4l2ctl.json

rtspServer: DEBUG:root:StreamServer.launch
INFO:root:USB camera ignored most of the parameters
DEBUG:root:( v4l2src is-live=true device=/dev/video0 brightness=50 contrast=50 saturation=50 ! queue ! jpegdec ! clockoverlay ! queue ! x264enc tune=zerolatency ! h264parse ! rtph264pay name=pay0 pt=96 )
INFO:root:Starting service on port 8554 at url /h264

rtspServer: INFO:root:Running RTSP Server

```

PapaGolf1957, 2023-01-14:
What version of Raspberry Pi OS are you using, I recently tried to get RPOS working on Bullseye and got similar errors to those you have reported. I then switched to Buster and it works ok.

enboig, 2023-01-14:
I am pretty sure it is Bullseye, I will check later

enboig, 2023-01-14:
I reinstalled Bullseye, gstreamer from apt, and continue failing:
```
rtspServer: ./python/gst-rtsp-launch.py:60: PyGIDeprecationWarning: GObject.MainLoop is deprecated; use GLib.MainLoop instead
  self.mainloop = GObject.MainLoop()
./python/gst-rtsp-launch.py:78: PyGIDeprecationWarning: Since version 3.11, calling threads_init is no longer needed. See: https://wiki.gnome.org/PyGObject/Threading
  GObject.threads_init()
INFO:root:StreamServer initialized

rtspServer: INFO:root:Video settings loaded from v4l2ctl.json

rtspServer: DEBUG:root:StreamServer.launch
INFO:root:USB camera ignored most of the parameters
DEBUG:root:( v4l2src is-live=true device=/dev/video0 brightness=50 contrast=50 saturation=50 ! queue ! jpegdec ! clockoverlay ! queue ! x264enc tune=zerolatency ! h264parse ! rtph264pay name=pay0 pt=96 )

rtspServer: INFO:root:Starting service on port 8554 at url /h264

rtspServer: INFO:root:Running RTSP Server

```

enboig, 2023-01-15:
> What version of Raspberry Pi OS are you using, I recently tried to get RPOS working on Bullseye and got similar errors to those you have reported. I then switched to Buster and it works ok.

Could you paste your .bash_history? I may do something different from you because it isn't working in my case

PapaGolf1957, 2023-01-19:
I have 2 Pis running Buster and PI HD cameras, I use the gsteamer RTSP source. These cameras have been running for some time as part of my house security. The record to a Network Video Recorder, connected via ONVIF. The RTSP output is  1920x1080 at 15 FPS. I used a 3rd PI recently using BULLSEYE and couldn't get it it to work. On Bullseye the camera  software uses Libcamera, thus the gst-rtsp-launch.py call to rpicamsrc fails. Not using a USB camera myself, my guess is there is a similar issue with v4l2src in gst-rtsp-launch not having the correct calls or libraries.

PapaGolf1957, 2023-02-08:
enboig
Sorry about the dely in getting back to you. I have now installed RPOS in a PI 4 running Bullseye, with a USB webcam. There are a number of issues install GSTREAMER and I am getting errors associated with gi.repository call in the gst-rtsp-launch.py. I am now stuck, but it does look as if there are a number of changes that are needed before RPOS is ready for bullseye. If I have time I will try an install of Buster.

PapaGolf1957, 2023-02-09:
enboig
Finally installed Buster and again ran into some issues, but the solution for me was to change the gst-rtsp-launch.sh file to remove all the bash code except for the line /usr/bin/python ./python/gst-rtsp-launch.py $@ -v. This assumes you are only running pytohn 2.7 and not python 3 (which fails similar to what you experienced). So I know think the issue with Bullseye lines around the fact it uses Python 3 by default.
Hope this helps

enboig, 2023-02-10:
It would help, but I will buy a second hand ip camera which already has ONVIF.

Thanks, I supose somebody with better python knowledge may upgrade the code.

closed 2023-02-10 by enboig