BreeeZe/rpos #156: Get exception from sax.js in node_modules

KianZhou8005, 2023-04-19:
Hi,

I have a Onvif client written by C++, it uses curl_easy_perform(curl) to query device info.

but it gets an response which shows below error:

Error: No whitespace between attributes Line: 0 Column: 114 Char: 
x     at error (D:\TEMP\rpos-master\node_modules\sax\lib\sax.js:651:10)
     at strictFail (D:\TEMP\rpos-master\node_modules\sax\lib\sax.js:677:7)     
at SAXParser.write (D:\TEMP\rpos-master\node _modules\sax\lib\sax.js:1395:13)     
at WSDL.xmlToObject (D:\TEMP\rpos-master\node_modules\soap\lib\wsdl.js:1381:5)     
at Server._process (D:\TEMP\rpos-master\node_modules\soap\lib\server.js:151:21)     
at IncomingMessage.<anonymous> (D:\TEMP\rpos-master\node_modules\soap\lib\server.js:124:14)     
at IncomingMessage.emit (node:events:513:28)     
at endReadableNT (node:internal/streams/readable:1359:12)     
at process.processTicksAndRejections (node:internal/process/task_queues:82:21) 

It seems the error is a XML converting issue, but I don`t know where is the XML and where is method is called

Can you give some advise?

RogerHardiman, 2023-06-26:
can you past the XML that you send out from your onvif client please?

RogerHardiman, 2023-08-07:
I stumbled across the same error myself two weeks ago and was able to find and fix it.
I also had to go looking for where the XML conversion is.

The cause is that there are characters after the ONVIF XML query.
So curl is sending something like this
<XML><blah blah blah></XML><null or whitespace><null or whitespace><null or whitespace>

it was NULL or whitespace after the end of the XML text that caused your crash.
It was the same for me.

So the current master branch has a fix for this.

It had to be fixed in an upstream library which RPOS forked (node-soap)

closed 2023-08-07 by RogerHardiman