BreeeZe/rpos #162: ONVIF to Visca Serial

nhparis, 2023-07-16:
Hi guys. It's not an "issue" per-se but I was wondering if there's any documentation regarding the PTZ Proxy or Protocol Convertor functionality you mention. Specifically, I would like to take an ONVIF joystick and control two Visca over serial PTZ cameras with it. The cameras are already linked to a Mac and work flawlessly (being controlled with Visca over serial from an OBS plugin) however I have recently been "donated" a top notch ONVIF (only) joystick. Does this software allow such conversion if I install it on the Mac? Thanks in advance
NH

RogerHardiman, 2023-07-16:
Hi
I don't know what a ONVIF Joystick is.
It is some kind of joystick with a small screen that can connect to ONVIF cameras? Do you have a picture or data sheet?

But you are correct, RPOS can work in a way where it receives PTZ commands in ONVIF and outputs VISCA.
The current version of RPOS only handles 1 physical camera so you have to run two instances of RPOS on different TCP Ports., I've got a development branch that supports multiple cameras via the one TCP Port but not merged in the code yet.

Thanks
Roger

nhparis, 2023-07-16:
Hi Roger. Yes that's exactly what it is. Specifically this one: https://www.amazon.co.uk/Hsility-Controller-Conference-Joystick-Decoding/dp/B088RK7VZR. It's normally used to control ONVIF compatible cameras directly over IP. However my cameras are not ONVIF compatible, so I'd like to find some way that the joystick sends ONVIF signals to [something - RPOS perhaps] and these are translated into VISCA via serial. My cameras (two like this https://communication.avereurope.com/model/cam540) are connected to an iMac via USB for receiving the video signal and two USB to serial converters for the VISCA PTZ control signal. If there was a way to install RPOS on the iMac and have it magically take the joystick IP signal and send it out over serial, that would be awesome! Thanks for any advice on how to do this... I'm a PC guy normally myself with a bit of RPi experience. Just saying :-)

RogerHardiman, 2023-07-16:
Hi
RPOS will run on a Mac. You need to install nodejs and npm on you Mac.
I used to run it on my Mac all the time until the Mac just got too slow (it was really old)
So you'd set up RPOS to use VISCA to a serial port in the rposConfig.json. For video you may want the 'testcard' branch which generates a bouncing ball video so you have something to 'view' on the joystick.

nhparis, 2023-07-16:
Oh and here's the user manual for the joystick if it's of any help. https://www.facebook.com/download/364335358283577/S3%20IP%20Controller--User%20Manual.pdf
And the Visca reference for the cameras.
https://communication.aver.com/DownloadFile.aspx?n=5168%7C4634A591-5862-4B45-A0E1-D0BC9AED855D&t=ServiceDownload

nhparis, 2023-07-16:
Awesome! I shall have a look and test tomorrow. Thanks! I'm sure installing nodejs and npm will be simple enough, but how and where exactly do I install RPOS? I apologise profusely.

nhparis, 2023-07-16:
Ignore that last comment... I've worked it out from the great documentation. I didn't realise it also walked through Mac install. Thanks again!

closed 2023-08-07 by RogerHardiman