BreeeZe/rpos #163: h265 encoded gstreamer

ozgurgunesulkesi, 2023-07-27:
Hi,

I wrote a python code creating a rtsp server using gstreamer.  I tryed rpos with it to make it onvif capable. Now the thing is tried two ways of encoding one h264 and one h265.  Now the thing is with h264 everything works fine but with h265 i have a bug when i open the onvif device on a NVR.  The thing is the stream also appears with h265. But when i change to fullscreen or do any other moving, change of resolution in the screen the stream turns black and does not appear. I encouter this only with h265.  Actually for encoding in gstreamer i am using x264enc or x265enc.

thats my launch string for gstreamer         self.launch_template= 'appsrc name=source is-live=True block=True format=GST_FORMAT_TIME ' \
                             'caps=video/x-raw,format={format},width={width},height={height},framerate={fps}/1 ' \
                             '! videoconvert ' \
                             '! video/x-raw,format=I420 ' \
                             '! x264enc tune=zerolatency ' \    #(or x265enc)
                             '! h264parse ' \  #(or h265parse)
                             '! rtph264pay name=pay0 pt=96 '   ##(or rtph265pay)

is this maybe related to my way i am encoding  or maybe with rpos itself?

RogerHardiman, 2023-07-28:
It is possible it is RPOS itself.
In the ONVIF Standard, your NVR will ask RPOS what the video format is.
It is possible the NVR then initialises itself (and its user interface) expecting H265 data. Perhaps it uses the reply from RPOS (an ONVIF reply) to initialise a hardware video decoder chip).
So what may be required is extending the RPOS code so it can reply to the NVR with an ONVIF reply that says that the video is H265.

However that is not a 1 line change.
The ONVIF Standards group implemented a whole new API (called 'Media2' which has to be used to tell the NVR that H265 is being used and RPOS does not have a Media2 service in the 'services' folder.

So there is a couple of days of work to get that working.

ozgurgunesulkesi, 2023-07-29:
Ok thank you for the fast reply. Actually i never wrote in javascript. So I will wait maybe until you update it.  thanks