BreeeZe/rpos #166: Big latency with usbcam

SyzkoAndrew, 2023-08-02:
Hello. I`m using your software for stream CSI and USB cameras. When i working with rpicam latency about few miliseconds. But when i use any usb camera latency is from 3 to 5 seconds. What can i change to make latency smaller such us bitrate, framerate or resolution? Can you help me resolve my porblem?

RogerHardiman, 2023-08-07:
Unfortunately I don't know how to reduce the USB latency. The original gstreamer code was written by someone @Schwaneberg Oliver Schwaneberg and although I have added bits to it, I don't know how to make the pipeline low latency.
I'm not sure if the latency is in the capture/encode stage or in the gstreamer RTSP server

SyzkoAndrew, 2023-08-07:
Thank you for a reply. Please not close issues a few time, about a week. Maybe someone could help me

closed 2023-11-09 by SyzkoAndrew