BreeeZe/rpos #167: ONVIF doesn't work in hikvision DVR

FelipeHurtado, 2023-09-11:
hi,
I am working with an IP camera which I retransmit RTSP streams with mediamtx, the Onvif protocol works well with Onvif device manager software but when trying to add ONVIF to the Hikvision DVR I get a "param error". I do not know what's happening

RogerHardiman, 2024-02-11:
Thank you for the bug report.
I have a Hik recorder, but don't remember if I've ever tried to make it connect to RPOS.

Could I ask
a) if you managed to fix the problem?
b) a copy of the RPOS config file please?
c) a copy of the MediaMTX config - if you have passwords in the MediaMTX file then please modify them

sbcdave, 2024-09-30:
Hey there. This is the newest "thread" about DVR issues so I'm hoping you don't mind if I tag along on this one, even though my issue is with a Reolink DVR. I am using rtsp-simple-server to proxy a Wyze cam rtsp stream, and then using RPOS to be the ONVIF front end for that proxy endpoint. Onvier and tinyCam can connect via ONVIF and see the stream, and my Reolink NVR autodiscovers and connects, and shows the frame rate and resolution info, but never displays the stream. On rtsp-simple-server output I see it "tearing down" the connection every minute or so. I noticed that the Reolink DVR sees the ONVIF "object?" as <Manufacturer>%20<Model>X where X is some unknown character. I'm hoping you can point me to where in code the manufacturer config value is being concatenated to the model config value, so I can modify it and see if that makes the darn Reolink DVR happy. Thanks for any pointers or other ideas. I really appreciate this RPOS code

sbcdave, 2024-09-30:
Nevermind the name having %20 and the weird character. I found where that was with a grep -r and then modified it to hardcode the name for me. The odd character was a \n immediately after the .concat(manfucaturer, model). The name now has no spaces or odd characters but it still isn't working. With log level 4 the NVR makes it all the way to the URI request, but stalls there until it destroys the connection and tries again. Any ideas are appreciated:

Discovery received from 192.168.68.58
DeviceService - Calltype : info, Data : Handling POST on /onvif/device_service
DeviceService - Calltype : received, Data : <?xml version="1.0" encoding="UTF-8"?>
<SOAP-ENV:Envelope xmlns:SOAP-ENV="http://www.w3.org/2003/05/soap-envelope" xmlns:SOAP-ENC="http://www.w3.org/2003/05/soap-encoding" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:wsa="http://schemas.xmlsoap.org/ws/2004/08/addressing" xmlns:wsdd="http://schemas.xmlsoap.org/ws/2005/04/discovery" xmlns:xop="http://www.w3.org/2004/08/xop/include" xmlns:chan="http://schemas.microsoft.com/ws/2005/02/duplex" xmlns:wsa5="http://www.w3.org/2005/08/addressing" xmlns:c14n="http://www.w3.org/2001/10/xml-exc-c14n#" xmlns:ds="http://www.w3.org/2000/09/xmldsig#" xmlns:saml1="urn:oasis:names:tc:SAML:1.0:assertion" xmlns:saml2="urn:oasis:names:tc:SAML:2.0:assertion" xmlns:wsu="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-wssecurity-utility-1.0.xsd" xmlns:xenc="http://www.w3.org/2001/04/xmlenc#" xmlns:wsc="http://docs.oasis-open.org/ws-sx/ws-secureconversation/200512" xmlns:wsse="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-wssecurity-secext-1.0.xsd" xmlns:tt="http://www.onvif.org/ver10/schema" xmlns:wsrfbf="http://docs.oasis-open.org/wsrf/bf-2" xmlns:wstop="http://docs.oasis-open.org/wsn/t-1" xmlns:wsrfr="http://docs.oasis-open.org/wsrf/r-2" xmlns:ns1="http://www.onvif.org/ver20/media/wsdl" xmlns:tdn="http://www.onvif.org/ver10/network/wsdl" xmlns:tds="http://www.onvif.org/ver10/device/wsdl" xmlns:tev="http://www.onvif.org/ver10/events/wsdl" xmlns:wsnt="http://docs.oasis-open.org/wsn/b-2" xmlns:timg="http://www.onvif.org/ver20/imaging/wsdl" xmlns:tmd="http://www.onvif.org/ver10/deviceIO/wsdl" xmlns:tptz="http://www.onvif.org/ver20/ptz/wsdl" xmlns:trc="http://www.onvif.org/ver10/recording/wsdl" xmlns:trp="http://www.onvif.org/ver10/replay/wsdl" xmlns:trt="http://www.onvif.org/ver10/media/wsdl" xmlns:trv="http://www.onvif.org/ver10/receiver/wsdl" xmlns:tse="http://www.onvif.org/ver10/search/wsdl" xmlns:ter="http://www.onvif.org/ver10/error" xmlns:tns1="http://www.onvif.org/ver10/topics" xmlns:tad="http://www.onvif.org/ver20/analytics/wsdl"><SOAP-ENV:Header><wsse:Security><wsse:UsernameToken><wsse:Username>admin</wsse:Username><wsse:Password Type="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-username-token-profile-1.0#PasswordDigest">LD6e0jbqZ/7Dnr17vmwIOv1MZPA=</wsse:Password><wsse:Nonce EncodingType="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-soap-message-security-1.0#Base64Binary">mwH7Zik++GFyqOVndqM3Tg1aIm8=</wsse:Nonce><wsu:Created>2024-09-30T19:52:59Z</wsu:Created></wsse:UsernameToken></wsse:Security></SOAP-ENV:Header><SOAP-ENV:Body><tds:GetCapabilities><tds:Category>All</tds:Category></tds:GetCapabilities></SOAP-ENV:Body></SOAP-ENV:Envelope>

DeviceService - Calltype : info, Data : Attempting to bind to /onvif/device_service
DeviceService - Calltype : info, Data : Trying Device from path 
DeviceService received request GetCapabilities
DeviceService - Calltype : replied, Data : <?xml version="1.0" encoding="utf-8"?><soap:Envelope xmlns:soap="http://www.w3.org/2003/05/soap-envelope"  xmlns:i0="http://www.onvif.org/ver10/device/wsdl" xmlns:tns="http://tempuri.org/" xmlns:tm="http://microsoft.com/wsdl/mime/textMatching/" xmlns:tds="http://www.onvif.org/ver10/device/wsdl" xmlns:tt="http://www.onvif.org/ver10/schema"><soap:Body><tds:GetCapabilitiesResponse xmlns:tds="http://www.onvif.org/ver10/device/wsdl" xmlns="http://www.onvif.org/ver10/device/wsdl"><tds:Capabilities><tt:Device xmlns:tt="http://www.onvif.org/ver10/schema"><tt:XAddr>http://192.168.68.54:8000/onvif/device_service</tt:XAddr><tt:Network><tt:IPFilter>false</tt:IPFilter><tt:ZeroConfiguration>false</tt:ZeroConfiguration><tt:IPVersion6>false</tt:IPVersion6><tt:DynDNS>false</tt:DynDNS><tt:Extension><tt:Dot11Configuration>false</tt:Dot11Configuration><tt:Extension></tt:Extension></tt:Extension></tt:Network><tt:System><tt:DiscoveryResolve>false</tt:DiscoveryResolve><tt:DiscoveryBye>false</tt:DiscoveryBye><tt:RemoteDiscovery>false</tt:RemoteDiscovery><tt:SystemBackup>false</tt:SystemBackup><tt:SystemLogging>false</tt:SystemLogging><tt:FirmwareUpgrade>false</tt:FirmwareUpgrade><tt:SupportedVersions><tt:Major>2</tt:Major><tt:Minor>5</tt:Minor></tt:SupportedVersions><tt:Extension><tt:HttpFirmwareUpgrade>false</tt:HttpFirmwareUpgrade><tt:HttpSystemBackup>false</tt:HttpSystemBackup><tt:HttpSystemLogging>false</tt:HttpSystemLogging><tt:HttpSupportInformation>false</tt:HttpSupportInformation><tt:Extension></tt:Extension></tt:Extension></tt:System><tt:IO><tt:InputConnectors>0</tt:InputConnectors><tt:RelayOutputs>1</tt:RelayOutputs><tt:Extension><tt:Auxiliary>false</tt:Auxiliary><tt:AuxiliaryCommands></tt:AuxiliaryCommands><tt:Extension></tt:Extension></tt:Extension></tt:IO><tt:Security><tt:TLS1.1>false</tt:TLS1.1><tt:TLS1.2>false</tt:TLS1.2><tt:OnboardKeyGeneration>false</tt:OnboardKeyGeneration><tt:AccessPolicyConfig>f [... cut ...] sdl" xmlns="http://www.onvif.org/ver10/media/wsdl"><trt:MediaUri><tt:Uri xmlns:tt="http://www.onvif.org/ver10/schema">rtsp://192.168.68.54:554/proxied</tt:Uri><tt:InvalidAfterConnect xmlns:tt="http://www.onvif.org/ver10/schema">false</tt:InvalidAfterConnect><tt:InvalidAfterReboot xmlns:tt="http://www.onvif.org/ver10/schema">false</tt:InvalidAfterReboot><tt:Timeout xmlns:tt="http://www.onvif.org/ver10/schema">PT30S</tt:Timeout></trt:MediaUri></trt:GetStreamUriResponse></soap:Body></soap:Envelope>