BreeeZe/rpos #168: Support for RPI Camera Module 3

jollywombats, 2023-12-15:
Good Evening,

Has anyone been able to get this to work with the RPI Camera Module v3 by chance?  Uncertain how to get it to work with that one as it just does not detect it, guessing due to the different driver library.

Thank you!

RogerHardiman, 2023-12-15:
I only have the original camera module so never tried a V3 module

sravich1, 2023-12-18:
Hi,
I am a trying to run the rpos code in Rasppi4 ..when I run the rpos.js I am getting following error
rtspServer error: Error: spawn ./bin/rtspServer ENOENT
at Process.ChildProcess._handle.onexit (internal/child_process.js:268:19)
at onErrorNT (internal/child_process.js:470:16)
at processTicksAndRejections (internal/process/task_queues.js:84:21) {
errno: 'ENOENT',
code: 'ENOENT',
syscall: 'spawn ./bin/rtspServer',
path: './bin/rtspServer',
spawnargs: [Array]
}

looks like rtsp server is unable to start...need to know what changes need to be made in addition to all the installation done.
Kindly let me know .
Thanks
Ravi

sravich1, 2024-02-09:
> I only have the original camera module so never tried a V3 module

I am trying with RPI4 ..facing issues..Kindly let me know if you have any suggestion 
Here is the following way in which I downloaded and started with the rpos

download from
git clone https://github.com/BreeeZe/rpos.git
cd rpos
npm install

do npx gulp
get the rpos.js
now use rposConfig.sample-picam.json to copy and create rposConfig.json..change the IP address here
"IpAddress" : "10.35.62.108",
4 "ServicePort" : 8081,
5 "Username" : "admin",
6 "Password" : "admin",
7 "CameraType" : "picam",
8 "RTSPAddress" : "", "//":"Normally left blank. Used to set RTSP Server Address",
9 "RTSPPort" : 8554,
10 "RTSPName" : "h264",
11 "MulticastEnabled" : false,
12 "RTSPMulticastName" : "h264m",
13 "MulticastAddress" : "224.0.0.1",
14 "MulticastPort" : "10001",
15 "RTSPServer" : 3, "RtspServerComment" : "## Select RTSP Server > 1:RPOS RTSP Server 2:V4L2 RTSP Server by mpromonet (auto selected if MulticastEnabled=true)",
I have used RTSPserver 3 here...with1 getting error

getting the ffmpeg error
ffmpeg exec error: Error: Command failed: ffmpeg -fflags nobuffer -probesize 256 -rtsp_transport tcp -i rtsp://127.0.0.1:8554/h264 -vframes 1 -r 1 -s 640x360 -y /dev/shm/snapshot.jpg
ffmpeg version 4.3.6-0+deb11u1+rpt5 Copyright (c) 2000-2023 the FFmpeg developers'
**getting mainly FFMPEG errors**

regards
Ravi

RogerHardiman, 2024-02-11:
Hi @jollywombats 
Is it possible to give me some remote access to a Pi with Camera Module 3 and then I can take a look?
Or did you get your problem sorted?

jollywombats, 2024-02-11:
@RogerHardiman    Sure.  Shoot me a PM and we can arrange something for this via ssh tunnel to the device.  

If you could get the cam3 module working + latest rasbiain version on a pi0w2 I will buy you the cam3 module :)