BreeeZe/rpos #169: trying to run rpos in raspberry pi 4

sravich1, 2023-12-19:
Hi,
I am newbie trying to run the rpos in raspberry pi 4.Not able to get the rtsp working.
Initially got the following errror
_rtspServer error: Error: spawn ./bin/rtspServer ENOENT
    at Process.ChildProcess._handle.onexit (internal/child_process.js:268:19)
    at onErrorNT (internal/child_process.js:470:16)
    at processTicksAndRejections (internal/process/task_queues.js:84:21) {
  errno: 'ENOENT',
  code: 'ENOENT',
  syscall: 'spawn ./bin/rtspServer',
  path: './bin/rtspServer',
  spawnargs: [Array]
}_

changed to option 3 in rposConfig.json...still not able to get the rtsp working..any method to run the same
_{
  "NetworkAdapters" : ["awdl0","eth0", "wlan0", "en0"],
  "IpAddress" : "10.35.62.102",
  "ServicePort" : 8081,
  "Username" : "admin",
  "Password" : "admin",
  "CameraType" : "picam",
  "RTSPAddress" : "10.35.62.102",
  "//": "Normally left blank. Used to set RTSP Server Address",
  "RTSPPort" : 8554,
  "RTSPName" : "h264",
  "MulticastEnabled"  : false,
  "RTSPMulticastName" : "h264m",
  "MulticastAddress"  : "224.0.0.1",
  "MulticastPort"     : "10001",
  "RTSPServer" : 3, "RtspServerComment" : "## Select RTSP Server > 1:RPOS RTSP Server 2:V4L2 RTSP Server by mpromonet (auto selected if MulticastEnabled=true)",
  "PTZDriver" : "none", "PTZDriverComment": "## valid values are none,tenx,pelcod,visca and pan-tilt-hat",
  "PTZOutput" : "none", "PTZOutputComment": "## values are none (eg Tenx), serial and tcp",
  "PTZSerialPort" : "/dev/ttyUSB0",
  "PTZSerialPortSettings" : { "baudRate":2400, "dataBits":8, "parity":"none", "stopBits":1 },
  "PTZOutputURL": "127.0.0.1:9999",
  "PTZCameraAddress": 1,
  "DeviceInformation" : {
    "Manufacturer" : "Raspberry Pi",
    "Model" : "4 B",
    "HardwareId" : ""
  },
  "logLevel" : 3, "logLevelComment": "## LogLevels are > 1:Error 2:Warning 3:Info 4:Debug",
  "logSoapCalls" : false_

Let me know if you have any chnages here.
Thanks
Ravi

universe241981, 2023-12-20:
What is the source of your rtsp stream? How is the rtsp path look like?
If your rtsp stream is not from picam and also not provided by V4L2 RTSP Server by mpromonet, then you have to use the "rposConfig.sample-proxy.json" template.

Also, the "RTSPAddress" : "hostname_of_remote_rtsp_server", "//":"Normally left blank. Used to set RTSP Server Address", should be at same line.

sravich1, 2023-12-22:
Hi,
My rtsp source is form pi cam as I am using raspiberry Pi 4 ...but still not able to get the rtsp server started.
I am working in one of the image where rtsp stream is working...but when I download and start afresh I am not able to get the stream...I even made the rtsp address same line and tried still not working.
regards
Ravi

sravich1, 2023-12-22:
current json
_{
  "NetworkAdapters" : ["awdl0","eth0", "wlan0", "en0"],
  "IpAddress" : "192.168.29.174",
  "ServicePort" : 8081,
  "Username" : "admin",
  "Password" : "admin",
  "CameraType" : "picam",
  "RTSPAddress" : "", "//":"Normally left blank. Used to set RTSP Server Address",
  "RTSPPort" : 8554,
  "RTSPName" : "h264",
  "MulticastEnabled"  : false,
  "RTSPMulticastName" : "h264m",
  "MulticastAddress"  : "224.0.0.1",
  "MulticastPort"     : "10001",
  "RTSPServer" : 3, "RtspServerComment" : "## Select RTSP Server > 1:RPOS RTSP Server 2:V4L2 RTSP Server by mpromonet (auto selected if MulticastEnabled=true)",
  "PTZDriver" : "none", "PTZDriverComment": "## valid values are none,tenx,pelcod,visca and pan-tilt-hat",
  "PTZOutput" : "none", "PTZOutputComment": "## values are none (eg Tenx), serial and tcp",
  "PTZSerialPort" : "/dev/ttyUSB0",
  "PTZSerialPortSettings" : { "baudRate":2400, "dataBits":8, "parity":"none", "stopBits":1 },
  "PTZOutputURL": "127.0.0.1:9999",
  "PTZCameraAddress": 1,
  "DeviceInformation" : {
    "Manufacturer" : "Raspberry Pi",
    "Model" : "2 B",
    "HardwareId" : ""
  },
  "logLevel" : 3, "logLevelComment": "## LogLevels are > 1:Error 2:Warning 3:Info 4:Debug",
  "logSoapCalls" : false
}_
where I have changed rtspserver to 3 but still gives following error
_rtspServer: /home/rasppi1/onvif/rpos/./python/gst-rtsp-launch.py:60: PyGIDeprecationWarning: GObject.MainLoop is deprecated; use GLib.MainLoop instead
  self.mainloop = GObject.MainLoop()

rtspServer: /home/rasppi1/onvif/rpos/./python/gst-rtsp-launch.py:78: PyGIDeprecationWarning: Since version 3.11, calling threads_init is no longer needed. See: https://wiki.gnome.org/PyGObject/Threading
  GObject.threads_init()

rtspServer: INFO:root:StreamServer initialized

rtspServer: INFO:root:Video settings loaded from v4l2ctl.json
DEBUG:root:StreamServer.launch
DEBUG:root:( rpicamsrc preview=false bitrate=10000000 keyframe-interval=60 drc=3 image-effect=denoise shutter-speed=0 iso=0 brightness=50 contrast=0 saturation=0 sharpness=0 awb-mode=1 rotation=0 hflip=False vflip=False video-stabilisation=False ! video/x-h264, framerate=30/1, width=1280, height=720 ! h264parse ! rtph264pay name=pay0 pt=96 )
INFO:root:Starting service on port 8554 at url /h264

rtspServer: INFO:root:Running RTSP Server_
if I give as 1 then error posted at the start comes...so basically RTSP server is not accessible.
Let me know if there needs to be any change.
regards
Ravi

universe241981, 2024-01-01:
1) Make sure the rtsp stream path is accessible. This can be done by open network stream via VLC Player.
I am using mediamtx as rtsp server for pi camera. It usually has rtsp stream path as "rtsp://<ip-address>:8554/stream" 

2) Rename the "rposConfig.sample-proxy.json" to "rposConfig.json" and change some parameter as below:

{
  "NetworkAdapters" : ["awdl0","eth0", "wlan0", "en0"],
  "IpAddress" : "192.168.1.15",
  "ServicePort" : 8081,
  "Username" : "admin",
  "Password" : "admin",
  "CameraType" : "picam",
  "RTSPAddress" : "", "//":"Normally left blank. Used to set RTSP Server Address",
  "RTSPPort" : 8554,
  "RTSPName" : "stream",
  "MulticastEnabled"  : false,
  "RTSPMulticastName" : "h264m",
  "MulticastAddress"  : "224.0.0.1",
  "MulticastPort"     : "10001",
  "RTSPServer" : 0, "RtspServerComment" : "## Select RTSP Server > 0:Use RTSP Server from 'RTSPAddress:' 1:RPOS RTSP Server 2:V4L2 RTSP Server by mpromonet (auto selected if MulticastEnabl>
  "PTZDriver" : "none", "PTZDriverComment": "## valid values are none,tenx,pelcod,visca and pan-tilt-hat",
  "PTZOutput" : "none", "PTZOutputComment": "## values are none (eg Tenx), serial and tcp",
  "PTZSerialPort" : "/dev/ttyUSB0",
  "PTZSerialPortSettings" : { "baudRate":2400, "dataBits":8, "parity":"none", "stopBits":1 },
  "PTZOutputURL": "127.0.0.1:9999",
  "PTZCameraAddress": 1,
  "DeviceInformation" : {
    "Manufacturer" : "RPiCam-3B",
    "Model" : "RPi3B",
    "HardwareId" : "1.0"
  },
  "logLevel" : 3, "logLevelComment": "## LogLevels are > 1:Error 2:Warning 3:Info 4:Debug",
  "logSoapCalls" : false
}

By the way, I have node v12.22.12 and npm v6.14.16.

sravich1, 2024-01-11:
Hi,
Thanks for your response.
1. I did try with mediamtx...but I am getting error 

NF [path cam] [RPI Camera source] ready: 1 track (H264)
2024/01/11 18:06:14 INF [RTSP] [conn 10.35.62.103:60852] opened
2024/01/11 18:06:14 INF [RTSP] [conn 10.35.62.103:60852] closed: invalid URL (/h264)
My IP is 10.35.62.102 but I do not know why it is picking up 103...also I installed from following path
_https://qengineering.eu/install-gstreamer-1.18-on-raspberry-pi-4.html_
If you have clear steps for this kindly provide
2.Once you run the mediamtx..then try with rposProxy changes...then run node rpos.js ....you are saying ONVIF complaint can be achieved??
can you ellaborate on this??
also directly with **onvif rpos** ..I cannot get the rtsp running

_rtspServer: /home/rasppi1/onvif/rpos/./python/gst-rtsp-launch.py:60: PyGIDeprecationWarning: GObject.MainLoop is deprecated; use GLib.MainLoop instead
  self.mainloop = _GObject.MainLoop()_

rtspServer: /home/rasppi1/onvif/rpos/./python/gst-rtsp-launch.py:78: PyGIDeprecationWarning: Since version 3.11, calling threads_init is no longer needed. See: https://wiki.gnome.org/PyGObject/Threading
  GObject.threads_init()

rtspServer: INFO:root:StreamServer initialized

rtspServer: INFO:root:Video settings loaded from v4l2ctl.json

rtspServer: DEBUG:root:StreamServer.launch
DEBUG:root:( rpicamsrc preview=false bitrate=10000000 keyframe-interval=60 drc=3 image-effect=denoise shutter-speed=0 iso=0 brightness=50 contrast=0 saturation=0 sharpness=0 awb-mode=1 rotation=0 hflip=False vflip=False video-stabilisation=False ! video/x-h264, framerate=30/1, width=1280, height=720 ! h264parse ! rtph264pay name=pay0 pt=96 )

rtspServer: INFO:root:Starting service on port 8554 at url /stream
rtspServer: INFO:root:Running RTSP Server_

**but not seeing anything in the vlc or any other streaming**
Kindly let me know the steps..I am in right path
Thanks and regards
Ravi

sravich1, 2024-01-16:
Hi , 
Any info on this error has any one tried running the same ...Kindly let me know for any pointers.
Thanks and regards
Ravi

sravich1, 2024-01-17:
Hi,
still getting ffmpeg issue on RPI4 or on compute module when running RPOS...

ffmpeg exec error: Error: Command failed: ffmpeg -fflags nobuffer -probesize 256 -rtsp_transport tcp -i rtsp://127.0.0.1:8554/h264 -vframes 1  -r 1 -s 640x360 -y /dev/shm/snapshot.jpg
ffmpeg version 4.3.6-0+deb11u1+rpt5 Copyright (c) 2000-2023 the FFmpeg developers
  built with gcc 10 (Debian 10.2.1-6)
  configuration: --prefix=/usr --extra-version=0+deb11u1+rpt5 --toolchain=hardened --incdir=/usr/include/aarch64-linux-gnu --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libdav1d --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librabbitmq --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sdl2 --disable-mmal --enable-neon --enable-v4l2-request --enable-libudev --enable-epoxy --enable-sand --libdir=/usr/lib/aarch64-linux-gnu --arch=arm64 --enable-pocketsphinx --enable-libdc1394 --enable-libdrm --enable-vout-drm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared
  libavutil      56. 51.100 / 56. 51.100
  libavcodec     58. 91.100 / 58. 91.100
  libavformat    58. 45.100 / 58. 45.100
  libavdevice    58. 10.100 / 58. 10.100
  libavfilter     7. 85.100 /  7. 85.100
  libavresample   4.  0.  0 /  4.  0.  0
  libswscale      5.  7.100 /  5.  7.100
  libswresample   3.  7.100 /  3.  7.100
  libpostproc    55.  7.100 / 55.  7.100

    at ChildProcess.exithandler (child_process.js:308:12)
    at ChildProcess.emit (events.js:314:20)
    at maybeClose (internal/child_process.js:1022:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:287:5) {
  killed: true,
  code: null,
  signal: 'SIGTERM',
  cmd: 'ffmpeg -fflags nobuffer -probesize 256 -rtsp_transport tcp -i rtsp://127.0.0.1:8554/h264 -vframes 1  -r 1 -s 640x360 -y /dev/shm/snapshot.jpg'
}
Kindly help me out on this ...
Thanks
Ravi

sravich1, 2024-01-24:
> 1. Make sure the rtsp stream path is accessible. This can be done by open network stream via VLC Player.
>    I am using mediamtx as rtsp server for pi camera. It usually has rtsp stream path as [...]

Thanks for the reply...I was able to get the mediamtx running on RPI4...Now I ran the rposConfig.JSON by changing the proxy.JSON...but I am not able to get the ONVIF going...I am getting **ffmpeg issue** 

_ffmpeg - starting
ffmpeg - finished
ffmpeg exec error: Error: Command failed: ffmpeg -fflags nobuffer -probesize 256 -rtsp_transport tcp -i rtsp://127.0.0.1:8554/stream -vframes 1  -r 1 -s 640x360 -y /dev/shm/snapshot.jpg
ffmpeg version 4.3.6-0+deb11u1+rpt5 Copyright (c) 2000-2023 the FFmpeg developers
  built with gcc 10 (Debian 10.2.1-6)
  configuration: --prefix=/usr --extra-version=0+deb11u1+rpt5 --toolchain=hardened --incdir=/usr/include/aarch64-linux-gnu --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libdav1d --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librabbitmq --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sdl2 --disable-mmal --enable-neon --enable-v4l2-request --enable-libudev --enable-epoxy --enable-sand --libdir=/usr/lib/aarch64-linux-gnu --arch=arm64 --enable-pocketsphinx --enable-libdc1394 --enable-libdrm --enable-vout-drm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared
  libavutil      56. 51.100 / 56. 51.100
  libavcodec     58. 91.100 / 58. 91.100
  libavformat    58. 45.100 / 58. 45.100
  libavdevice    58. 10.100 / 58. 10.100
  libavfilter     7. 85.100 /  7. 85.100
  libavresample   4.  0.  0 /  4.  0.  0
  libswscale      5.  7.100 /  5.  7.100
  libswresample   3.  7.100 /  3.  7.100
  libpostproc    55.  7.100 / 55.  7.100
[rtsp @ 0x5586036000] method DESCRIBE failed: 404 Not Found
rtsp://127.0.0.1:8554/stream: Server returned 404 Not Found_

any pointers for this error....I am running both the 2 different windows/sessions..let me know 
--Thanks
Ravi

sravich1, 2024-01-24:
why is RTSP method failed coming from....the method in which I am running is fine right??...kindly help me out.
--Ravi

sravich1, 2024-01-24:
My ffmpeg conversion from .mp4 to mkv is working fine...do not know why specifically I am getting this error..
any changes need to be made in ffmpeg.??

sravich1, 2024-01-24:
when setup for ONVIF is clicked ..ffmpeg is giving the error...any idea where the problem is ??

universe241981, 2024-01-27:
1. Make sure you can play the rtsp stream by using VLC player or similar software.

2. If your rtsp stream need username and password in order for it to play?

3. `ffmpeg exec error: Error: Command failed: ffmpeg -fflags nobuffer -probesize 256 -rtsp_transport tcp -i rtsp://127.0.0.1:8554/stream -vframes 1 -r 1 -s 640x360 -y /dev/shm/snapshot.jpg`
Try change the rtsp path to the "absolute" rtsp path. If the rtsp need username and password, please supply that into the rtsp path also.

universe241981, 2024-01-27:
The rtsp path gonna be edit here "rpos/services
/media_service.ts" before compile them from .ts to .js.

sravich1, 2024-02-08:
Hi,
Thanks for your reply...
Here is the following way in which I downloaded and started with the rpos
1. download from 
git clone https://github.com/BreeeZe/rpos.git
cd rpos
npm install

2. do npx gulp
get the rpos.js 
now use rposConfig.sample-picam.json to copy and create rposConfig.json..change the IP address here
 "IpAddress" : "10.35.62.108",
 4   "ServicePort" : 8081,
 5   "Username" : "admin",
 6   "Password" : "admin",
 7   "CameraType" : "picam",
 8   "RTSPAddress" : "", "//":"Normally left blank. Used to set RTSP Server Address",
 9   "RTSPPort" : 8554,
10   "RTSPName" : "h264",
11   "MulticastEnabled"  : false,
12   "RTSPMulticastName" : "h264m",
13   "MulticastAddress"  : "224.0.0.1",
14   "MulticastPort"     : "10001",
15   "RTSPServer" : 3, "RtspServerComment" : "## Select RTSP Server > 1:RPOS RTSP Server 2:V4L2 RTSP Server by mpromonet (auto selected if MulticastEnabled=true)",
**I have used RTSPserver 3 here**...with1 getting error
4. getting the ffmpeg error
 _**ffmpeg exec error: Error: Command failed: ffmpeg -fflags nobuffer -probesize 256 -rtsp_transport tcp -i rtsp://127.0.0.1:8554/h264 -vframes 1  -r 1 -s 640x360 -y /dev/shm/snapshot.jpg
ffmpeg version 4.3.6-0+deb11u1+rpt5 Copyright (c) 2000-2023 the FFmpeg developers**_

Now any pointers let me know where the errors are so that I can make the changes 
thanks 
Ravi

sravich1, 2024-02-08:
Hi ,
I tried downloading mediamtx and running rtsp...RTSP is running successfully.
but when I change the rposConfig.json under rpos I have this issue.
That's the update.

RogerHardiman, 2024-02-11:
When you are running MediMTX you need to
a) stop RPOS from launching a RTSP server
    This requires "RTSPServer": 0

b) make sure MediaMTC knows the RTSP URL for the video stream
So if your MediaMTX is on the same computer as RPOS and if, for example, the MediaMTX URL was rtsp://hosrname:8554/stream
then you would set the folloiwng
"RTSPAddress" : "_put the IP address of the device running Media MTX here_"
"RTSPPort" : 8554,
"RTSPName" : "stream

You can try with "RTSPAddress": "127.0.0.1" but you probably want to put the actual IP address because that RTSPAddress forms part of FFMPEG being used to grab 1 frame of video and make a JPEG and also because ONVIF returns the RTSP URL back to the VMS or CCTV viewing software which will then be told to get video from 127.0.0.1 and not from a real IP address

RogerHardiman, 2024-02-11:
I should try and add an RTSPServer option 4 some time to make it launch MediaMTX automatically

universe241981, 2024-02-19:
> rtsp://10.35.62.108:8554/cam

If your rtsp stream working (verify by play it in VLC Player) with the above path, then in the rposConfig file, the "RTSPName" : "h264" has to change to "RTSPName" : "cam".

By the way, is the rtsp link below working as "rtsp://admin:admin@10.35.62.108:8554/cam"? 
Or "rtsp://10.35.62.108:8554/cam" should work?
If "rtsp://10.35.62.108:8554/cam" is working, then in rposConfig file, the  "Username" : "" & "Password" : "" gonna leave as is...don't put username and password on it.

sravich1, 2024-02-19:
Hi ,
Thanks for all the suggestion ..mediamtx for RTSP and RPOS for ONVIF worked in raspberry pi
[MediaMtx_RPOS_stepstoinstall.txt](https://github.com/BreeeZe/rpos/files/14330970/MediaMtx_RPOS_stepstoinstall.txt)

Thanks for all your help.
--ravi