BreeeZe/rpos #172: Waveshare pan-tilt-hat on pi4 communication through onvif in frigate (hassio)

berndklug1107, 2024-06-08:
Hi everybody!

my objective: detection and autotracking with ov5647 picam on waveshare pt-hat on pi4 / 4GB or pi zero2 using rpos onvif

this is a user-specified issue, so afaik no bug or similar because regarding the guides and docs it should work, but I'm not 100% sure.

symptom:
the pt-hat (servos) works with the common waveshare or other py library when calling the scripts on pi4 or zero, so everything works locally. during installation (buster and bullseye) with the mounted pt-equipment apart from warnings everything runs ok and finally the rpos-onvif interface works, getting rtsp-stream form mediamtx (instead of the rpos rtsp streamer) and serves the streams in frigate(hassio) hacs integration, installed on another pi5, so the cams work in frigate with onvif protocol except the pt-control :-(
I'm not sure what the rpos ptz-driver needs to address the pt-servos, if some kind of (waveshare) py-script, controlling the servos, needs to run in order to handle communication with rpos ptz-driver (like that: https://github.com/RogerHardiman/node-pan-tilt-hat ) or if the rpos ptz driver itself communicates with the pt-hat, this I also found somewhere, but can't find anymore :-)

Running hawkeye217/fovtest.py script gives this:

YES - GetServiceCapabilities shows that the camera supports MoveStatus.
NO - MoveStatus not reporting IDLE or MOVING.
NO - RelativeMove Pan/Tilt is unsupported.
NO - RelativeMove Zoom is unsupported.

So maybe someone has a similar configuration (must be :-) and give me a hint 🙏🙏🙏

attached config & log-files
[frigate-log.txt](https://github.com/user-attachments/files/15749360/frigate-log.txt)
[rposCOnifg.json](https://github.com/user-attachments/files/15749361/rposCOnifg.json)
[rpos-install.txt](https://github.com/user
[frigate.yml.txt](https://github.com/user-attachments/files/15749394/frigate.yml.txt)
-attachments/files/15749362/rpos-install.txt)

RogerHardiman, 2024-09-05:
I have a Waveshare PT HAT and it was on my TODO list to add support alongside the Pymoroni Pan/Tilt HAT but I never had time to do the work.

To answer your question, RPOS will turn any RTSP stream into an ONVIF camera with PTZ support.
RPOS receives the PTZ commands and sends them on to whatever motors are supported.
In the RPOS source is a typescript file that handles the ONVIF PTZ messages and supports a wide range of motor controls - Pymoroni Pan/Tilt HAT, or output in classic Pelco D format for old CCTV hardware, or an ASCII protocol etc.
So you will need to dig into that source code and work out how to forward the motor controls on to the Waveshare Pan/Tilt HAT

RogerHardiman, 2024-09-05:
Oh, turns out I did implement Waveshare support and it is in this alternative project...
     https://github.com/RogerHardiman/node-pan-tilt-driver

It was 6 years ago, I'd forgotten.

So take a look at this project. It probably helps.

berndklug1107, 2024-09-05:
Thank you!
Yes, I already discovered the waveshare node-pan-tilt-driver link, so I easily replaced the existing /home/pi/rpos/node_modules/pan-tilt-hat folder and it works perfectly 😍 ... of course after modifying the hat hardware itself, this is a complete mess 🙄
I was little confused because of all the different statements and opinions regarding the cams itself, but meanwhile I know what it's all about