BreeeZe/rpos #174: How Build for windows ?

masgh021, 2024-10-08:
Hello @BreeeZe 
How run onvif server in windows ?

RogerHardiman, 2024-10-09:
most of this project will run fine with NodeJS on Windows.
git clone the source code
then run "npm install" and "npx gulp" and you should have a rpos.js file you can run.

The built in RTSP servers are Linux specific so you will need to use your own RTSP server on Windows like MediaMTX

I used to develop RPOS on Windows most of the time. Now I use WSL on Windows

masgh021, 2024-10-25:
Hello @RogerHardiman 
i test this command npm install and got this error , can you help me ?
[image]

traeu, 2024-12-21:
Unfortunately I'm also unable to try this out on Windows 10.
I don't have any experience with node.js.
Installed it from nodejs.org, also installed "additional tools for node.js".
After downloading the repo and running "npm install", I get the following result:

<details>
<summary> CMD Output </summary>

```

C:\Users\FCS Research Lab\Downloads\rpos>npm install
npm warn skipping integrity check for git dependency ssh://git@github.com/BreeeZe/node-soap.git
npm warn deprecated source-map-resolve@0.6.0: See https://github.com/lydell/source-map-resolve#deprecated
npm warn deprecated urix@0.1.0: Please see https://github.com/lydell/urix#deprecated
npm warn deprecated har-validator@5.1.5: this library is no longer supported
npm warn deprecated source-map-resolve@0.5.3: See https://github.com/lydell/source-map-resolve#deprecated
npm warn deprecated chokidar@2.1.8: Chokidar 2 does not receive security updates since 2019. Upgrade to chokidar 3 with 15x fewer dependencies
npm warn deprecated resolve-url@0.2.1: https://github.com/lydell/resolve-url#deprecated
npm warn deprecated source-map-url@0.4.0: See https://github.com/lydell/source-map-url#deprecated
npm warn deprecated node-uuid@1.4.8: Use uuid module instead
npm warn deprecated uuid@3.4.0: Please upgrade  to version 7 or higher.  Older versions may use Math.random() in certain circumstances, which is known to be problematic.  See https://v8.dev/blog/math-random for details.
npm warn deprecated request@2.88.2: request has been deprecated, see https://github.com/request/request/issues/3142
npm error code 1
npm error path C:\Users\FCS Research Lab\Downloads\rpos\node_modules\@serialport\bindings
npm error command failed
npm error command C:\Windows\system32\cmd.exe /d /s /c prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild
npm error serialport.cpp
npm error C:\Users\FCS Research Lab\Downloads\rpos\node_modules\nan\nan_callbacks.h(55,23): error C2039: "AccessorSignature" ist kein Member von "v8". [C:\Users\FCS Research Lab\Downloads\rpos\node_modules\@serialport\bindings\build\bindings.vcxproj]
npm error   (Quelldatei „../src/serialport.cpp“ wird kompiliert)
npm error       C:\Users\FCS Research Lab\AppData\Local\node-gyp\Cache\22.12.0\include\node\v8.h(71,11):
npm error       Siehe Deklaration von "v8"
npm error
npm error C:\Users\FCS Research Lab\Downloads\rpos\node_modules\nan\nan_callbacks.h(55,23): error C2065: "AccessorSignature": nichtdeklarierter Bezeichner [C:\Users\FCS Research Lab\Downloads\rpos\node_modules\@serialport\bindings\build\bindings.vcxproj]
npm error   (Quelldatei „../src/serialport.cpp“ wird kompiliert)
npm error
npm error C:\Users\FCS Research Lab\Downloads\rpos\node_modules\nan\nan_callbacks.h(55,13): error C2923: „v8::Local“: „AccessorSignature“ ist kein gültiges Vorlage-Typargument für Parameter „T“. [C:\Users\FCS Research Lab\Downloads\rpos\node_modules\@serialport\bindings\build\bindings.vcxproj]
npm error   (Quelldatei „../src/serialport.cpp“ wird kompiliert)
npm error       C:\Users\FCS Research Lab\Downloads\rpos\node_modules\nan\nan_callbacks.h(55,23):
npm error       Siehe Deklaration von "AccessorSignature"
npm error
npm error C:\Users\FCS Research Lab\Downloads\rpos\node_modules\nan\nan_callbacks.h(55,42): error C2955: "v8::Local" : Für die Verwendung von Klasse Vorlage ist eine Vorlage-Argumentliste erforderlich [C:\Users\FCS Research Lab\Downloads\rpos\node_modules\@serialport\bindings\build\bindings.vcxproj]
npm error   (Quelldatei „../src/serialport.cpp“ wird kompiliert)
npm error       C:\Users\FCS Research Lab\AppData\Local\node-gyp\Cache\22.12.0\include\node\v8-local-handle.h(252,22):
npm error       Siehe Deklaration von "v8::Local"
npm error
npm error C:\Users\FCS Research Lab\Downloads\rpos\node_modules\nan\nan.h(2516,14): error C2955: "v8::Local" : Für die Verwendung von Klasse Vorlage ist eine Vorlage-Argumentliste erforderlich [C:\Users\FCS Research Lab\Downloads\rpos\node_modules\@serialport\bindings\build\bindings.vcxproj]
npm error   (Quelldatei „../src/serialport.cpp“ wird kompiliert)
npm error       C:\Users\FCS Research Lab\AppData\Local\node-gyp\Cache\22.12.0\include\node\v8-local-handle.h(252,22):
npm error       Siehe Deklaration von "v8::Local"
npm error
npm error C:\Users\FCS Research Lab\Downloads\rpos\node_modules\nan\nan.h(2516,34): error C2641: Für "v8::Local" können keine Vorlagenargumente abgeleitet werden. [C:\Users\FCS Research Lab\Downloads\rpos\node_modules\@serialport\bindings\build\bindings.vcxproj]
npm error   (Quelldatei „../src/serialport.cpp“ wird kompiliert)
npm error
npm error C:\Users\FCS Research Lab\Downloads\rpos\node_modules\nan\nan.h(2516,34): error C2780: „v8::Local<T> v8::Local(const v8::LocalBase<T> &)“ erwartet, dass 1 Argumente – 0 angegeben werden. [C:\Users\FCS Research Lab\Downloads\rpos\node_modules\@serialport\bindings\build\bindings.vcxproj]
npm error   (Quelldatei „../src/serialport.cpp“ wird kompiliert)
npm error       C:\Users\FCS Research Lab\AppData\Local\node-gyp\Cache\22.12.0\include\node\v8-local-handle.h(401,3):
npm error       Siehe Deklaration von "v8::Local"
npm error
npm error C:\Users\FCS Research Lab\Downloads\rpos\node_modules\nan\nan.h(2516,34): error C2780: „v8::Local<T> v8::Local(const v8::Local<T> &,v8::api_internal::StackAllocated<false>::no_checking_tag)“ erwartet, dass 2 Argumente – 0 angegeben werden. [C:\Users\FCS Research Lab\Downloads\rpos\node_modules\@serialport\bindings\build\bindings.vcxproj]
[... cut ...]
npm error gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\FCS Research Lab\\AppData\\Roaming\\npm\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
npm error gyp ERR! cwd C:\Users\FCS Research Lab\Downloads\rpos\node_modules\@serialport\bindings
npm error gyp ERR! node -v v22.12.0
npm error gyp ERR! node-gyp -v v11.0.0
npm error gyp ERR! not ok
npm error A complete log of this run can be found in: C:\Users\FCS Research Lab\AppData\Local\npm-cache\_logs\2024-12-21T16_32_54_733Z-debug-0.log

```

</details>

Is there a way to run this on Windows 10?
Or any other known projects that can wrap a RTSP-stream in ONVIF?