BreeeZe/rpos #175: Can I use this to turn a remote rtsp stream into an onvif stream?

latel, 2024-11-06:
I searched a lots, this project seems to be the closest implemention, but currently it can only stream from pi local cameras or local stream files.

can I stream from a remote rtsp stream, and turn it into an onvif stream? My viewer only supports onvif cameras.

latel, 2024-11-06:
yes, it's possible, take following config as a sample
```
{
  "NetworkAdapters" : ["awdl0","eth0", "wlan0", "en0"],   -> will determine onvif proxy address
  "IpAddress" : "", -> your remote rtsp server address
  "ServicePort" : 18880,  -> onvif proxy port
  "Username" : "",  -> onvif proxy username
  "Password" : "",  -> onvif proxy password
  "CameraType" : "picam", -> anything is fine
  "RTSPAddress" : "10.10.10.11",  -> anything is fine
  "RTSPPort" : 9554, -> your remote rtsp server port
  "RTSPName" : "tapo_c120", -> your remote rtsp server stream path
  "MulticastEnabled"  : false,
  "RTSPMulticastName" : "h264m",
  "MulticastAddress"  : "224.0.0.1",
  "MulticastPort"     : "10001",
  "RTSPServer" : 0, -> must select 0
  .........
}

```

closed 2024-11-06 by latel

sbcdave, 2024-11-25:
Hi there. I have that same config. My Reolink NVR sees the camera, but it loops through ONVIF calls and never displays the stream. Are you able to help me troubleshoot this by any chance?

latel, 2024-11-26:
NVR typically support rtsp cameras as well, there is no need to wrap it as a onvif camera.

sbcdave, 2024-11-26:
Mine unfortunately does not. There are two options when you add a camera, IP or ONVIF. In either case you provide the IP, port, username, and password, but there is no place to specify the path portion of the URI. The NVR is able to connect to my RPOS ONVIF server, and it gets the manufacturer, model, location, ect...data, but when it gets to requesting the URI it hangs connects to the RTSP IP:port, but hangs for some reason, for about 30 seconds, and then disconnects and starts over. I've tried everything I can think of to make my RPOS replicate what one of their Reolink cameras does, but I'm missing something and the Reolink NVR isn't working. Any ideas are appreciated.